wled/WLED #411: UDP sync issues to multiple ESP8266’s

betadeltadelta, 2019-12-03:
Excellent software!!! I have an issue I can’t sort out though with the built in UDP sync. I’ve looked over setting several dozen times and I’m so stumped!!  I have 5 units with static IP’s 201 through 205 on their own wireless network, not connected to the internet or anything else.

These lights syncs excellently with Xlights, but I don’t think I will always run those sequences. Therefore, I’d like to use built in presets I’ve made on 201, my front door and set that one to sync to others. The front door is the one I have presets saved to and a macro to run them at bootup/powerup. For the most part it’s good, but the unit to the left of my door and sometimes the far right are lazy and don’t receive the update/make the change. 

Other times they go through 2-3 presets and then get stuck again...

Any ideas?
[image]
[image]
[image]
[image]

betadeltadelta, 2019-12-03:
[image]
[image]
[image]

betadeltadelta, 2019-12-03:
Also these are my snowflakes I built. This video is running with xlights for proof of concept before I hung them outside. I can give more detail if needed but they are all identical. 126 ws2812b and 94 ws2811’s Over my front door. Again, each has its own esp8266.

[WLED on custom 126 Pixel Flakes](https://youtu.be/qr0-_CTK7UI)

Aircoookie, 2019-12-04:
Hi! Wow, your christmas lights look amazing!! Really honored that that is all running off WLED!
About the sync, the commands sometimes not getting sent is something that might happen in some networks, especially if the router doesn't support a feature called IGMP. You can try to mitigate the issue by selecting "Send notifications twice" in Sync settings. Let me know if it helps!

betadeltadelta, 2019-12-04:
Thank you!! I do have that checked, yes. And would that matter if the issue was just one of the 5 nodes not receiving? That’s the oddball part. I will look for that router setting. I bought a brand new TP-Link router just for my display and so far, so great other than this issue. I feel like it’s just something with that node because sometimes it shows not available in the app. Like maybe it’s a connectivity issue. I will report back and do some better videos before the holiday is over linking to your github. It’s really awesome and very versatile software with so many features built in.

betadeltadelta, 2019-12-04:
Well I've looked into the above suggestion and it appears as though my router doesn't have any mention of the feature.  Either that or because I'm using in AP mode, since there is no internet connection.  I did notice something odd, maybe this will jog the minds to figure out.  I have them all on a timer and my front door unit is the master sending macros.  I noted that several other units had the sync switch "ON"!  Not sure if that stays off after reboot/powerdown or not?  My main units macro is "&T=1&A=127&CY=1&P1=1&P2=4&PT=15000&TT=1000"

Do you see anything that might be causing this issue with that macro at boot time?  It's turning them all on, setting brightness, cycling presets 1-4 along with duration and transition time.  Seem like maybe this could be an issue to send that macro to all devices when they don't have built in presets.  Trying to find the best way to wrap my head around this since there is no time/clock for the units, just a wall timer turning them on firing at boot.

ironosity, 2019-12-08:
I don't have a solution..but where's your instructions for building the snowflakes!? Those are sweet!

betadeltadelta, 2019-12-09:
@ironosity thank you!  No instructions other than I sketched them on autocad with the proper spacing considering IP65 LED strips at 30led/m.  I bought 4 rolls of them and cut them to appropriate lengths.  Each flake uses 126 leds, 21 per arm.  Used the sticky back to adhere to the flakes that were cut out of 5mm plywood from Lowe's and spray painted white.  Then, lots and lots of soldering the strips in sequential order so the signal follows the same pattern around each flake.   I soldered all power at the center so each and every arm has power injection.  That's also where my NodeMCU plugs in via an extension cord I made to house the nodes about 3' away sitting in my windowsill in a PVC pipe enclosure.  I could probably make a brief video if interested but I think what I just explained covers it.

betadeltadelta, 2019-12-09:
Also... Trying numerous things and reflashing and flashing different versions.... No matter what I try, that flake is lazy.  Works totally fine on E1.31... But WLED UDP packets will not cooperate, so I hesitate to say it's a bumb ESP8266.  Anybody?

ironosity, 2019-12-09:
I have 3 esp8266 on my porch and stairs, all syncing from a 4th on the roof lights. (I know, wierd, but I had data issues crossing the stairs and just gave up and put in controllers for each side of stairs. :D )
I will say that it's not abnormal for one of the lights to get stuck on a FX for 2 or more cycles and then it suddenly syncs again. It IS always the same one and it's the same distance from wifi as the other one. So I don't know what's going on. I often can't connect via app either so I assume it's just a flaky connection and so it misses the UDP packets.  It sucks. I kinda wish these boards had ethernet ports.

betadeltadelta, 2019-12-09:
@ironosity yeah what you describe sounds like my situation.  It doesn't not work, it's just very flaky.  Sometimes it will go through 2 or 3 5 second transitions and then the fourth it stays stuck on that effect for like 5 minutes.  Also, same as you, sometimes that unit isn't available via the app, it drops off connection somehow.  My dedicated wifi router is just in side the front door an there are only these 5 esp8266's and now a raspberry pi running FPP on it.  I don't understand why that one unit has the hickups.  Odd.

closed 2019-12-19 by betadeltadelta