wled/WLED #412: Flashes, something similar to #230

knochenkopp, 2019-12-03:
Yesterday I saw something strange. I installed about 15 meters of WS2812B IP65 strips in aluminium channels on my house.
They are controlled by a D1 Mini with WLED v0.8.5 on 70A power supply with a logic shifter. I have power injection about every 3-4 meters. The cable from thelogic shifter to the first Strip is about 1,5m in leght and arrives at one corner on the first strip, goes up the gable to the ridge, then down again (about 200 LEDs) and from there around the corner and horizontally along the house (about 300 LEDs). I had problems at this corner with a loose connection. So I divided the strip at this corner into two single strips that i end up with one on the gable and one horizontally along the house . I put a seperate cable to the corner and connected the horizontal strip directly to the D1 mini. That works perfectly. 
My electronics and power supply are in a waterproof box in the attic. Because I wanted  not always have to walk back and forth while testing things, so  I made a short "teststrip" of about 1 meter in length (about 30 LEDs) abd connectet it next to the box in the attic in parallel to the horizontal one with only about 10cm cable. That worked wonderfully, both lit up with the same pattern, no errors. 
Now I connected the first strip on the gable also in paralel. The three strips then all showed the same pattern. The funny thing was that now the teststrip and the strip on the gable began to flash randomly between the patterns, but the horizontal one dosen't show any flashes at all! I then disconnected the gablestrip, but the teststrip continued to flash, no flashes on the horizontal one. The same will appear if i disconnect the teststrip, then the gablestrip flashes.  The horizontal one again shows no flashes in any of the setups.
When I switched off the power supply off and turned it back on all was back to normal. The test strip and the horizontal light up again normally with the gablestrip disconnected. When I the reconnected the strip on the gable to the other two I again got the flashes immediately on the teststrip and the gablestrip, the horizontal one again shows no flashes. 
Why are the flashes appear not on all connected strips? And if the gablestrip is the problem, why does the flashing on the teststrip continues when I disconnect the gablestrip? And i doubt that the cable leght and a voltage drop of the data signal is the problem, because the strip that dosen't flash has about 6,5 meters from the logic shifter to the first LED and the flashing strips have 10cm and 1,5 meter. But maybe the flashing appears BECAUSE of the short cables? I'm mildly confused...

Aircoookie, 2020-01-05:
Hi, sorry for taking ages to get back to you on this!

When you connect multiple strips in parallel, they have to share the current the D1 and/or levelshifter can provide, which lowers the voltage and can induce flickers. Additionally, different types of LEDs (maybe the teststrip is from a different batch than the horizontal of gable) are more or less tolerant to changes on the data wire. What I would recommend is to get one pixel from the type of LEDs that flickers the least, hooking that up to the D1 output and then running the parallel outputs from the output of that one LED. You can also skip that LED using the option in the bottom of LED settings or segments :)

tonyn0, 2020-01-05:
You could also connect D1 to two or more level-shifter inputs and then break out the strips across those outputs.

closed 2020-05-11 by stale[bot]