wled/WLED #414: Feature Request: Dawn to Dusk timer

fijit586, 2019-12-04:
As many uses of the wled app involve use at night. I would like to see a dawn to dusk feature built in. Nodemcu is already connected to internet so couldn't it pull local dawn and dusk times and allow for auto on and off features. Enhanced by dusk + X hours (2,4,6,8) for customization.
I'm sort of newbie to this stuff so maybe this feature isn't executable in wled. If so, please forgive my ignorance on how this all works.

woody2010, 2019-12-14:
I use an amazon smart plug for this functionality since I don't want my 60amp power supply to run when not in use.

fijit586, 2019-12-14:
Thats exactly what i did. Alexa controlled and has schedule function. No power to psu or leds when not in use.

Regards,
Fred Tangelder
248-953-5122

closed 2020-04-19 by stale[bot]