wled/WLED #415: Enhancement: Merry Christmas FX too short

fijit586, 2019-12-04:
Hello, been having fun using wled to control my PERMANENT holiday leds. Due to the time of year the Merry Christmas FX was one of the first I tried. The secondary color, in this case green, has only 1 pixel/led light up on the red background. Because of how I have this diffused in my windows, the green is barely noticeable. I would love the ability to define the "length" of the secondary color. That is to say how many sequential pixels turn the secondary color ex..(1,2,4,6) would be perfect. 
I'm also guessing many of the other FX could benefit from a "length" slider control, just like they benefit from a "Speed" control.
Keep up the great work.
P.S. Eagerly awaiting second data channel support for NodeMCU.

ironosity, 2019-12-04:
Merry Christmas already lights up to 6 LEDs red/gree based on how you set the Intensity slider.

fijit586, 2019-12-04:
hello eric, thanks for the quick response.
I am running v8.5.
I changed the intensity slider several times. There is no change in the ratio of grn to red pixels (1:3).

Regards,
Fred Tangelder
248-953-5122

ironosity, 2019-12-04:
oh...I didn't know it was a recent addition! I started on 8.6 and am now running latest compiled. Good news is it will be in .9! :)

betadeltadelta, 2019-12-04:
Maybe try the running command, set red as primary and green as secondary color.  Mess with the sliders to see if that simulates what you're trying to accomplish.  Sounds like it will be close.

fijit586, 2019-12-04:
Thanks, that actually works well.

Regards,
Fred Tangelder
248-953-5122

closed 2020-01-05 by Aircoookie