wled/WLED #417: add ArtNet support

daavery, 2019-12-05:
it would be nice if WLED supported ArtNet in addition to E1.31.

artnet can be either very simple - ArtNetDMX packet only or complex with support for RDM and remote programming of node features

I'm looking at various artnet code bases there a couple for esp but none are async :
https://github.com/rstephan/ArtnetWifi
https://github.com/natcl/Artnet
https://github.com/mtongnz/ESP8266_ArtNetNode_v2
https://github.com/hideakitai/ArtNet

there is a  more complete codebase for reference:
https://github.com/OpenLightingProject/libartnet

the spec for it all is:
https://art-net.org.uk/resources/art-net-specification/

thanks

Aircoookie, 2019-12-05:
Thank you for finding me a few links. Looking at https://github.com/rstephan/ArtnetWifi, it seems to communicate via UDP, just like E1.31 and be quite straightforward. If we are lucky, I should be able to add some nice async Artnet support!

abyssdj, 2020-02-07:
What is the progress on this so far? Great job on the software, it is much better than alternatives I have been using!

T-Arens, 2020-04-10:
Wanted to ask the same, is there any code for Artnet yet? Because otherwise I might give it a try. :-) I just used ESP8266_ArtNetNode_v2 on an ESP-01 to get my [RGBW LED Moving Head](https://www.lixada.com/p-l0761us.html) connected via WiFi, and that works beautifully with the free PC software [DMXControl](https://www.dmxcontrol.org). It has a nice editor for creating custom devices, too. Now I'd like to integrate my WLED based sideboard illumination, and here I am, thinking about Artnet. :-)

pille, 2020-04-10:
initially i wanted to have that, too. i'm now hooked on E1.31, which was elementary supported and improved that. since there's currently a lot going on (refactoring, open DMX pull requests) i would postpone this a little bit. the idea is to implement ArtNet eventually alongside the E1.31 implementation, as they are just envelopes for the same letter.

Aircoookie, 2020-04-10:
@T-Arens decided I want to work on this today. If everything works out, we might have artnet support in a few hours!

Aircoookie, 2020-04-10:
@pille yep, I'm just extending the `ESPAsyncE131` to deal with the different ArtNet packet structure. Will just call `handleE131Packet()` and work exactly the same as E1.31 as far as WLED is concerned.

abyssdj, 2020-04-11:
Artnet would be awesome. The reason it works better in my case is that I can have multiple devices subscribed to the same universe, which my software automatically configures to unicast. SACN on the other hand will only let you set up a single device to each universe in unicast, and multicast is of course not ideal on wifi.

pille, 2020-04-11:
@abyssdj: i'm using E1.31 in multicast mode on WIFI and have no problems. as all lights share the same packet, this actually reduces bandwith on the network.

Aircoookie, 2020-04-12:
WLED now supports Artnet in latest master source! Feel free to give it a try, it needs to be enabled in Sync settings and then rebooted.

T-Arens, 2020-04-14:
Nice, thx! 
It doesn't work though, I think because DMXControl first scans all Artnet devices with "[ArtPoll packet (0x2000)](https://art-net.org.uk/structure/discovery-packets/artpollreply/)" (from Wireshark) and does not send out anything as long as there is no answer.

abyssdj, 2020-04-14:
Just compiled and uploaded, appears to be working. Not tried with two or more devices yet, so will update if that raises any issues. However, the computer (Lightjams) is not detecting and displaying the node so this must be working in multicast mode. As @pille suggested, on paper this should use less bandwidth I guess, but either way it would be ideal if it was detectable by other nodes. I expect this is the same as @T-Arens refers to above.

Aircoookie, 2020-04-14:
Yep, WLED currently implements code from hideakitai's project, which only supports ArtDmx packets and as such not the polling discovery mechanism. natcl's library seems to implement the ArtPoll mechanism  though, that could ease porting it to WLED :)

abyssdj, 2020-04-14:
That explains that then. Either way though, I'm finding it much more responsive and fluid than SACN.

T-Arens, 2020-04-15:
Thanks for highlighting, Aircookie. Do you plan to integrate ArtPoll into WLED by any chance?

Aircoookie, 2020-04-15:
I am considering that, would definitely be useful! But it may take a little while until i have time for it

abyssdj, 2020-12-15:
Evening! Just noticed this has been changed to ArtPoll and bumped to 1.0.0, is there any way it can be implemented quicker? Would it work from a usermod, in some way?

Aircoookie, 2020-12-15:
Sorry, I do have other features that are requested more at the time, so they are higher up on the priority list. Feel free to give it a try yourself or find someone to add it, I'd be happy about a PR :)

WerkAG, 2020-12-18:
I discovered this project less than a month ago, and more and more I'm excited with it. Just for fun, yesterday I tried WLED with LEDFX using E1.31, it worked like a charm. Rapidly I started to think how nice it could be using a professional lighting desk to generate the effects, and control WLED via Artnet... well, the lack of response to ArtPoll make me stop, as the WLED node isn't discovered.
So, just for info, I also would like to have ArtPoll support, it will greatly extend the use possibilities. However, I can wait until 1.0.0
Once again congratulations for this great project.

schleyma, 2021-02-07:
@Aircoookie : This project is a blast - Thanks so far!
I was wondering if anyone has a clue why I can't get Art-Net running when sending data in broadcast-mode. 
I've been successfully using DMX-Workshop for years to test my Art-Net components - for whatever reason I get my WLEDs only to react to data when sending Art-Net data manually to WLED's IP address. I played around with the option MULTICAST but this does not seem to make any difference...

Glad if someone has an idea...

Thanks
Matt

dzelionis, 2022-04-10:
Hey i have work around, check my post https://github.com/Aircoookie/WLED/issues/2256

erhan-, 2022-05-01:
I can confirm that it is working with @dzelionis added layer currently but would also be happy if it worked natively. Thanks for everyhing already done. I love WLED!

abyssdj, 2022-05-01:
> I can confirm that it is working with @dzelionis added layer currently but would also be happy if it worked natively. Thanks for everyhing already done. I love WLED!

I had forgotten about this, I had a very quick look some time ago but didn't have time to try it. If I remember right, we still had to manually input the IP addresses for the devices we wanted to advertise via artnet, is that still the case, or does it automatically detect and advertise wled nodes?

abyssdj, 2022-05-01:
Replying to myslef - 

I've just compiled and tested this on my office ceiling lights, Lightjams discovers them and correctly names them in the software

https://github.com/4lloyd/WLED

https://github.com/Aircoookie/WLED/pull/2615

erhan-, 2022-05-01:
Yes, I have to add the WLEDs IP addresses to an array in his script. Thank you very very much for your PR and contribution!

abyssdj, 2022-05-01:
> Yes, I have to add the WLEDs IP addresses to an array in his script. Thank you very very much for your PR @4lloyd and contribution! Thanks for hinting to this @abyssdj. I have overseen it!

No problem, I hadn't seen it either but so far, working perfectly. Lightjams currently discovering three different led strings immediately and assigning them all unicast IP addresses.

dzelionis, 2022-05-02:
I agree, would be nice if that would work natively. Maybe, someday, Aircoookie will find time to get it completed. Until then, we still can survive... :D Enjoy.

abyssdj, 2022-05-02:
> I agree, would be nice if that would work natively. Maybe, someday, Aircoookie will find time to get it completed. Until then, we still can survive... :D Enjoy.

Well, as this is already working as expected I am sure it will be merged soon enough!

erhan-, 2022-05-02:
Well I though compiling would be a good solution until I found out that I flashed an OTAless version onto a novostella bulb which means that I am stuck on the current version forever now :)

dzelionis, 2022-05-19:
Have new version of the script which addresses all existing issues......
https://github.com/dzelionis/-artnet_advertise

erhan-, 2022-05-27:
@abyssdj I have compiled this for me and at least Soundswitch does still not detect it. With @dzelionis script it still works. It would still be better to remove one step inbetween. I think @dzelionis sends the artnet message without a poll first or so, then soundswitfch detects it.

dzelionis, 2022-05-28:
Agree, would be nice to have wled working out of box with soundswitch or
other Artnet applications. If you look into artnet documentation, you will
find that device has to advertise itself every 2.5 -3 seconds.

erhan-, 2022-05-30:
Wait wait, I have to backpaddle. I wanted to start my lights and powered the two I flashed on while soundswitch is on. They went blank, which means they connected. I checked the hardware tab and they are indeed there with the right names. I am not sure what I made different. 

It works!

erhan-, 2022-05-31:
I just compiled this PR and flashed it on all devices. It works quite good but there was a moment where none was detected somehow. But might be more an issue on my end or maybe I was using the Artnet port at that moment.

dzelionis, 2022-05-31:
kk, i will try also to compile and flash it

dzelionis, 2022-06-01:
the first attempt failed, not detected by soundswich on pc&prime :/

dzelionis, 2022-06-01:
just to confirm, i have compiled and flashed this one:
https://github.com/4lloyd/WLED

i have port mirroring setup on my switches, running wireshark captures and I expect to see some packets for ptotovol Artet  Pool/PoolReply from 10.4.20.29 (ESP32 with wired Lan) and 10.4.20.14 (ESP32 wifi)....see none ...
[image]

if i start my script:
[image]

erhan-, 2022-06-01:
Have your checked out the artnet branch before you compiled?

They had their device descriptions as names. Yours sends MyArtnet so I was able to confirm that way, that at least that information is coming indeed from those devices.

dzelionis, 2022-06-10:
> Have your checked out the artnet branch before you compiled?
> 
> They had their device descriptions as names. Yours sends MyArtnet so I was able to confirm that way, that at least that information is coming indeed from those devices.
> 
> edit: Annddd doesnt work again :) Not sure what the magic trick is. Works with your script.
> 
> edit 2: Dont ask me why but it works with the prime controller. It detects the lights correctly with the right names.

Hey, 
had no chance to do that, still need to find some time to try again....
will update when i have another atempt.

dzelionis, 2022-06-25:
Hey, quick update,
so this tme i have compiled the correct reopo, 
and yes its working, I was testing it with soundswirch software running on denon prime4.
i did notice that sometimes it disappears from arnter device list on soundswitch for a few secs....(may need more rapid pool updates on wled to fix that) - but it did not caused any major issues...so, so far so good.....

@Aircoookie any plans to integrate it to main repo ?

Also, one thing that is a bit annoying me, that there is no fast way to diasble/ignore DMX input on wled if you are running it in DMX Effect mode. So lets say my player stopped playing and will start sending all 0 as DMX data, so i can't just go on my phone and light wled...everytime i need to go to setting, disable DMX, save, and only then it usable again....but i guess i need to open new feature request for it...

blazoncek, 2022-06-25:
Hi,

As Aircoookie may be absent until further notice, please summarise the issue in a few sentences as I do not have the time to digest everything posted here.
I may take a look if the summary is well written.

dzelionis, 2022-06-25:
> Hi,
> 
> As Aircoookie may be absent until further notice, please summarise the issue in a few sentences as I do not have the time to digest everything posted here. I may take a look if the summary is well written.

Hi, @blazoncek . In that case, let's just focus on the Topic of the issue. Basically, WLED does not have ArtNet ArtPoll built in support, so that means multiple applications would just not detect wled device via ArtNet Protocol, to resolve that currently, we have two options: the first one was use third-party application (like my script in python) to do a job or we can compile not official wled which also works fine ( i played arround around 1week+ with it). Link to it: **https://github.com/4lloyd/WLED/tree/add-artpoll**

So, merging "add-artpoll" repo with the main official one would resolve this ticket.

Thanks
dzelionis

blazoncek, 2022-06-25:
I assume you are talking about #2615 (and possibly #2619).
In this case it has already been decided that it **will** be merged **after** #2610.

Unfortunately #2610 is such a big PR that needs thorough testing by Aircoookie. Unfortunately, as I said earlier, he cannot do that ATM and I do not want to rush him or merge without approval.

I hope things will get better soon and we'll be back on track.
Meanwhile use your hack or the code @4lloyd provided.

dzelionis, 2022-06-25:
> I assume you are talking about #2615 (and possibly #2619). In this case it has already been decided that it **will** be merged **after** #2610.
>
> Unfortunately #2610 is such a big PR that needs [...]

Sound like a plan!
Thanks!

davidd1205, 2022-07-04:
very good job with soundswich unstable on 8266 but very stable on esp32 with prime 4 
thanks for your job

davidd1205, 2022-07-11:
@dzelionis 
Hello
I also use wled with the firmware that integrates the poll for artnet, it works well in effect mode, in sounswitch you can choose the effects with the cues that's not bad but not easy in autoscript because you need a cue to stop the effects also suddenly I'm thinking of programming in static looks and launching them manually finally it's my choice I don't know if you do the same or not, there may be a trick that I haven't found, however I have 2 questions concerning it integration of wled in soundswitch, the first is that you have noticed that below 20% intensity, the leds go out, whereas when controlling them with the wled ui live, I don't have this problem but SS sends too low an intensity value, it seems, did you notice? moreover SS created 2 children by peripheral wled you know why? is it for multirgb mode or others? I did not understand how to operate a mode with several segments none of my tests were good, thank you for your work,
I also saw that you had a prime 4 with a modified firmware, what did you add or improve on the prime4, thank you

dzelionis, 2022-07-13:
> Thank you for your answer, it's nice to share.
> 
> yes the auto loops are nice but I also use autoscript for my playlists it goes even further for wled and sounswitch I'm not seeing with Liam from SS but he asks me for the documentation of the wled as for a "normal" fixture I don't really know what to provide him lol
> 
> I'll let you know if we move forward with Liam
> 
> I tried your effect with the cue attributes in SS the one you show in your screenshot with the explanations for the various settings to be made in wled and SS it is really nice I search the effects by passing the presets one by one but not easy if you have other settings (preset) to share I'm interested because I found the list of presets but I'm not sure it really corresponds thank you for the info on the prime 4 another big job on your part, I hope that the beta firmwares will bring us a little more functionality and freedom to take advantage of this touch screen a little more see you soon

for liam is not to much you need to provide,
all what he need is on this page:
https://kno.wled.ge/interfaces/e1.31-dmx/

Aircoookie, 2022-09-22:
ArtPoll should now be supported in the `0_14` branch :)

closed 2022-09-22 by Aircoookie

dzelionis, 2022-10-11:
That's, no problem at all,
enjoy it!
Best,
dzelionis