wled/WLED #419: 0.90.0 dev - new UI can't scroll effects page in safari on ios

tube0013, 2019-12-05:
With the new dev build of 0.9.0 on mobile on iOS 13.2.3 I'm not able to scroll the list of effects. I checked via the WLED ios app, Safari and the Chrome app.

[image]

Aircoookie, 2019-12-05:
Thank you, good that you found this issue! Seems like the function for the horizontal swipe cancels the scroll event (which doesn't happen on android). I hope I can find a solution quickly!

HaasNL2, 2019-12-10:
I noticed this is also in on Andriod.

Patrick-Plueard, 2019-12-14:
I just downloaded the latest release .9 1 hr ago and I am 
having vertical scrolling issues with Android.

closed 2020-04-20 by stale[bot]

SputnikSeven, 2021-03-16:
I'm not sure if it's the same bug, but this has been happening for me as well.  I can scroll the effects list down, but if I try to scroll it up it a circular type of wait symbol appears momentarily then it returns me to the main Colors screen. I'm on Android 10 on a Oneplus 6T and I have included a photo of the symbol that appears as scrolling. 
[image]