wled/WLED #420: wled fails to build in arduino ui

daavery, 2019-12-05:
fx.h
line 558 

    segment _segments[MAX_NUM_SEGMENTS] = { // SRAM footprint: 21 bytes per element
      // start, stop, speed, intensity, palette, mode, options, color[]
      { 0, 7, DEFAULT_SPEED, 128, 0, DEFAULT_MODE, NO_OPTIONS,{DEFAULT_COLOR}}

needs to be 

    segment _segments[MAX_NUM_SEGMENTS] = { // SRAM footprint: 21 bytes per element
      // start, stop, speed, intensity, palette, mode, options, color[]
      { 0, 7, DEFAULT_SPEED, 128, 0, DEFAULT_MODE, NO_OPTIONS,0,0,0, {DEFAULT_COLOR}}

daavery, 2019-12-05:
comment fixed:
      // start, stop, speed, intensity, palette, mode, options, group, spacing, opacity, color[]
      { 0, 7, DEFAULT_SPEED, 128, 0, DEFAULT_MODE, NO_OPTIONS,0,0,0, {DEFAULT_COLOR}}

Aircoookie, 2019-12-05:
Thank you! I don't know why it built for me before. Probably the compiler having a different level of strictness when it comes to variable initialisations.

closed 2019-12-05 by Aircoookie