wled/WLED #421: Multiple controller extension

ThomasHop8, 2019-12-05:
Hi all! 👋🏼 

First of all, I would really like to thank all the contributors and especially Aircoookie for this great project!

I have been using WLED already but want it to be a bit bigger this year. My current setup is as follows; I currently have multiple LED strips connected to multiple WLED instances. There are currently connected and synched together via UDP. Really happy with this setup, but I still have a question about a possible feature.

My question is if it is possible to use a WLED node to extend a LED strip. For example; I currently have the _“wipe”_ effect enabled with the synched option enabled on 2 controllers, but this applies the effect on both the strips. 

I would like to have the second controller to be an extension of a first controller so the behaviour will be; starting the _“wipe”_ effect so the effects starts only on one end of one LED strip and continues down the line all the way to the other end of the second LES strip. 

Is this something that is possible? And if so, how could I configure this? Unfortunately, in my current setup, achieving this via hardware extension is not possible, so I am looking for a software one. 

Thanks anyway, and keep up the good work!

stockklauser, 2019-12-05:
i think its realy hard to sync it over Wifi. You dont know the time that the package need to be transmittet over wifi.

But a possible way could: 
1: Use a Common NTP Server and set a Start time like 12:31:10 (hh:mm:ss) at every stripe + the time that the effect need to come to the end of the stripe (handover point) 
--- Example: The effect need calculatet 13 Seconds, its Starts on the First Stripe @ 13:31:10 then the second stripe can start at 13:31:43. 

2: We can add a Hardwire to sync (easy way)

Aircoookie, 2019-12-05:
Hi, thank you for the suggestion :)

I believe this is a very good idea! It has already been suggested here: https://github.com/Aircoookie/WLED/issues/338
Therefore I am closing this one if it's alright :)

@stockklauser yes you are right, network delay isn't high with UDP sync (typically below 100ms), but that can be enough to be noticable with a fast Wipe. Since the delay is often quite constant though, I believe adding a fine tune setting would be enough. The NTP implementation WLED uses is only accurate to the second, so not good enough for syncing.

closed 2019-12-05 by Aircoookie