wled/WLED #424: Dropping Wifi Connection

ozzi91, 2019-12-05:
Hello!
I have problem with drop wifi connection. After 20-30mins my wled on wemos d1 mini is not reachable till wemos reset. After reset wled is working again for next 20-30mins and goes off. I tried 2 wemos, and it is not working at all.

I tried wled_0.8.6_ESP8266_lwip1 and WLED_0.8.5_ESP8266 flashed by esphome flasher but it doesnt help-the same result. Distance between router and esp its like 10cm. I can provide you any information you want to check it out. If it is matter i integrate it with HA.

thedannymullen, 2019-12-08:
Have you tried without any leds on it?  I actually had this problem and found my led was causing the wemos to lock up.  I can’t explain it without a scope but assume it was pulling the pin down and causing a reset.  
I found another led and put one in series with my string as a null pixel.  Everything works fine.  
It was confirmed to be hardware as other software had the same issue with the suspect led tied in first.

ozzi91, 2019-12-09:
Hello! Ty for answer. I tried wit h2 differents leds set and even without any led. Still wifi droping. Blue led on wemos is on so wemos is working :/

ksuclipse, 2019-12-15:
i'm actually running into the exact same thing.  it failed right at 20 minutes though the effects are still running.  it just dropped off wifi.  here is json/info:

```{"ver":"0.8.6","vid":1910255,"leds":{"count":519,"rgbw":false,"pin":[2],"pwr":0,"maxpwr":65000,"maxseg":10},"name":"WLED Light","udpport":21324,"live":false,"fxcount":83,"palcount":50,"arch":"esp8266","core":"2_5_2","freeheap":18296,"uptime":1200,"opt":119,"brand":"WLED","product":"DIY light","btype":"bin"

markusn1, 2019-12-15:
I have the same issue with 0.8.6 and 0.9.0-b1. 
0.8.5 does not seem to be affected by this.
Could it be the "non-Wemos" D1 minis from aliexpress?

ksuclipse, 2019-12-15:
i was seeing it in 8.5 as well.  so additional testing, i had the esp8266 running standalone with no led's attached and it still dropped its wifi eventually.  Strangely when i ran it with AP mode set to always it ran all night with no issue however when i plugged it back into the LEDs it failed after about an hour.  Even the AP is no longer broadcasting.

markusn1, 2019-12-15:
I just tried using binaries which i compiled myself with different core versions (i tried 2.4.1 and 2.6.2), but it did not change anything.

ksuclipse, what board are you using?

ksuclipse, 2019-12-15:
Its a knock off nodemcu.  so i was wrong it is still working if i have the AP always on.  give it a try and see if i'm crazy

markusn1, 2019-12-20:
I tested all WiFi modes (on, on when not connected, off) to no avail. Still the same behavior for me. After about 10-20 mins the Wemos loses WiFi. But like i said, it is still connected to my accesspoint. If i kick it, it will come back reachable, but eventually loses connection again.

raoulteeuwen, 2019-12-20:
@Aircoookie Could the people at https://github.com/esp8266/Arduino/issues/5083 be on to something re ESP8266 and wifi / sleep? Or this one: https://github.com/esp8266/Arduino/issues/5912#issuecomment-478389370 ? This: https://community.home-assistant.io/t/esp8266-weird-wifi-issues/50484/26 ? This: https://forum.micropython.org/viewtopic.php?t=6860 ?

Aircoookie, 2019-12-21:
@raoulteeuwen Thank you for the links! I believe I will add a setting to enable `WiFi.setSleepMode(WIFI_NONE_SLEEP);`, but not enable it by default.

The only thing I do not like about it is that it triples the average power consumption of the ESP from 20mA to 60mA (which will also make it run hotter), but if it helps fix disconnects, that tradeoff would be worth it in those cases. I have a very flaky WLED installation myself (because it is outside, signal strength very low, around -80), I will test tomorrow if disabling sleep could help in that case.
All my indoor devices with good signal strength never seem to have any connectivity issues.

It could also be router-dependent, ESPs tend to run better with FritzBoxes or Netgear than TP-link or Asus units as far as i've heard.

Aircoookie, 2019-12-21:
An update on this. Setting sleep mode unfortunately doesn't seem to help with connectivity on my outside unit. I seems like the signal quality is just really poor on that unit (maybe hardware defect). I also opened the WLED-AP and I can only access it from outside, phone doesn't discover it inside. I will try and flash 0.8.5 (since many of you report issues since 0.8.6) just in case, but I do believe it is just because of the signal. It's a shame too since my phone has good Wifi reception at the place the outside unit is set up...

ozzi91, 2019-12-23:
Just for info, i really have no idea why this is working or not. I tried:
Use 3 different NODEMCU bought on Aliexpress.
Use 3 different Wemos D1 mini bought on alie
Turn it on without ir with led strip or led lights and it doesnt help.

Devices are like 10cm away from router (Netgear with openwrt). I used different power source aswell like usb form pc or extended power source.
I dont know what i can try more.

raoulteeuwen, 2019-12-23:
Could you try to assign one or more controllers (NodeMCU/Wemos) a fixed IP and see whether that has any influence? Just wondering whether any other router config option could be in play...

ksuclipse, 2019-12-23:
Mine is set as static ip, ap always on and connected to WiFi. Been
connected for about a week and it’s still online.

ozzi91, 2019-12-31:
Even with static ip problem still exist :/

markusn1, 2019-12-31:
> @raoulteeuwen Thank you for the links! I believe I will add a setting to enable `WiFi.setSleepMode(WIFI_NONE_SLEEP);`, but not enable it by default.
>
> The only thing I do not like about it is that it [...]

I am running a lot of ESP-based devices (Sonoff, Shelly, WemosD1 etc) on my WiFi (Ubiquiti Unifi UAP-AC-LR), and all are very stable, with the exception of D1s running WLED 0.8.6 and later. I had similar issues with Tasmota when it was compiled with the wrong ESP Core. Trying different core versions on newer versions of WLED did not help though.

kabelux, 2020-01-03:
My NodeMCU ESP32 is really wifi unstable. Keeps disconnecting every 2 or 5 minutes. I'm using latest beta version (glitter effect is so cool!)

Pocke01, 2020-01-08:
I have the same problem with a WemosD1 on a QuinLED-dig-uno and running wled 0.8.6. 

After a few minutes I'm unable to ping the ip and connect to the web interface.  
The strange thing is I'm still able to turn leds on/off with home assistant. 

I have tried with both dhcp and static ip.   If I use dhcp my server show the wled as offline at the same time as I'm unable to connect to the ip.  Im using Unifi AP-AC Pro for my wifi.

AnshumanFauzdar, 2020-01-20:
Ohh well, I think problem is not solved yet
I am also using Wemos D1 mini and same wifi connection drops after 15-20 minutes and I have to manually power off/on wemos!
Please do update if anyone finds a solution!
Thanks :)

paularmstrong, 2020-01-20:
I have the same issue with an ESP32s: #517. It's even more noticeable with that, as it often fails to stay on for even a few seconds. When it does come on, it'll stay up for ~15 minutes before it drops and needs power-cycling

AnshumanFauzdar, 2020-01-20:
> I have the same issue with an ESP32s: #517. It's even more noticeable with that, as it often fails to stay on for even a few seconds. When it does come on, it'll stay up for ~15 minutes before it drops and needs power-cycling

Lets see, if anybody comes up to a solution, WLED UI is very dope, I want to use this only!

markusn1, 2020-01-24:
I just compiled wled from the latest master branch, and so far it seems to be stable (around 2 hours so far). Will keep you updated!

AnshumanFauzdar, 2020-01-24:
> I just compiled wled from the latest master branch, and so far it seems to be stable (around 2 hours so far). Will keep you updated!

Ohh thank you brother, You did not used the binary?

markusn1, 2020-01-25:
So, around 20 hours later it is still running fine.
@AnshumanFauzdar yes i did not use one of the precompiled binaries, but compiles it from the most recent source.
@Aircoookie whatever you did, it seems to work for me!

AnshumanFauzdar, 2020-01-25:
Thanks brother, will compile it ASAP

sirs2k, 2020-01-27:
> I just compiled wled from the latest master branch

I'm sorry, still a noob here but having wifi issues like everyone else. 
Could you link me to the specific release for the esp8266?

Thank you

AnshumanFauzdar, 2020-01-27:
> > I just compiled wled from the latest master branch
> 
> I'm sorry, still a noob here but having wifi issues like everyone else.
> Could you link me to the specific release for the esp8266?
> 
> Thank you

Hey,
Have you complied from [here?](https://github.com/Aircoookie/WLED/tree/master/wled00)

Apapra, 2020-01-29:
My problem is that my four esp8266 nodemcu works perfectly with WLED 0.9.0-b2 (and static IP) till router reboot. I found that after reboot esp8266 don't reconnect to wifi till power off-on. Any way to fix this? Thanks!

AnshumanFauzdar, 2020-01-29:
> WLED 0.9.0-b2 

Where is b2 released?

Apapra, 2020-01-29:
> Where is b2 released?

It's not released, it's compiled from source code.

AnshumanFauzdar, 2020-01-29:
> It's not released, it's compiled from source code.
Okay, Thanks!

Apapra, 2020-01-29:
> My problem is that my four esp8266 nodemcu works perfectly with WLED 0.9.0-b2 (and static IP) till router reboot. I found that after reboot esp8266 don't reconnect to wifi till power off-on. Any way to fix this? Thanks!

Well, after reading various threads I think I should try to play with "Try connecting before opening AP for x s" setting.

AnshumanFauzdar, 2020-01-29:
> Well, after reading various threads I think I should try to play with "Try connecting before opening AP for x s" setting.

Please do update here if you are able to solve the problem
Thanks :)

Apapra, 2020-01-30:
I didn't find this setting in current firmware. :) But after I filled all fields in static IP section (IP, gateway, subnet) it seems that problem is solved. Now after router reboot nodemcu is available (online) again.

AnshumanFauzdar, 2020-01-30:
> I didn't find this setting in current firmware. :) But after I filled all fields in static IP section (IP, gateway, subnet) it seems that problem is solved. Now after router reboot nodemcu is available (online) again.

Okay thanks!

deepak-varade, 2020-04-17:
> An update on this. Setting sleep mode unfortunately doesn't seem to help with connectivity on my outside unit. I seems like the signal quality is just really poor on that unit (maybe hardware [...]

Hi
i think finally found exact solution for continuous rebooting of esp.
if i set time Time and Macros setting using any ntp server with my time zone.is working fine.No connection issue with esp server.Correct me if i am wrong.

AnshumanFauzdar, 2020-04-18:
> > An update on this. Setting sleep mode unfortunately doesn't seem to help with connectivity on my outside unit. I seems like the signal quality is just really poor on that unit (maybe hardware defect). I also opened the WLED-AP and I can only access it from outside, phone doesn't discover it inside. I will try and flash 0.8.5 (since many of you report issues since 0.8.6) just in case, but I do believe it is just because of the signal. It's a shame too since my phone has good Wifi reception at the place the outside unit is set up...
> 
> Hi
> i think finally found exact solution for continuous rebooting of esp.
> if i set time in Time and Macros setting using any ntp server with my time zone.is working fine.No connection issue with esp server.Correct me if i am wrong.

Thank you for your effort and solution
I just upgraded to latest firmware and now connection is not dropping at all
Thank you!

deepak-varade, 2020-04-18:
> Thank you for your effort and solution
> I just upgraded to latest firmware and now connection is not dropping at all
> Thank you!

Hi 
wled android app showing offline esp ,also when try to connect from Google chrome browser its getting frequently disconnected any idea what may be the problem?

AnshumanFauzdar, 2020-04-18:
> Hi
> wled android app showing offline esp ,also when try to connect from Google chrome browser its getting frequently disconnected any idea what may be the problem?

Try to reflash your hardware, it will solve your problem

gegu, 2020-04-22:
First, thanks for the great project.

For years I have had problems with all projects where a web server is started by ESP8266/ESP32.
Sooner or later I stopped the projects, because none of them worked stable for a long time. No matter if own projects, Tasmota, or whatever.

All settings, tips and recommendations unfortunately did not bring any improvement.
Now I have tested WLED - the same game again.

But now I could find a way to make the webserver stable at least in my environment, which I don't want to withhold from you.

- WLED configuration with static IP
- Every 60 seconds a ping to the local IP address of ESP.

**Implementation**:

Install library [ESP8266Ping](https://github.com/dancol90/ESP8266Ping)

_usermod.cpp_:
```cpp
#include "wled.h"
/*
 * This file allows you to add own functionality to WLED more easily
 * See: https://github.com/Aircoookie/WLED/wiki/Add-own-functionality
 * EEPROM bytes 2750+ are reserved for your custom use case. (if you extend #define EEPSIZE in const.h)
 * bytes 2400+ are currently ununsed, but might be used for future wled features
 */

// climits to define MAX_INT which is used by ESP8266Ping
#include <climits>
// ping lib
#include <ESP8266Ping.h>

// delay between pings
const int CheckWiFiDelayMs = 60000;
// WiFi connection status
bool connectedWiFi = false;
// last ping
long lastPingTime = 0;

//Use userVar0 and userVar1 (API calls &U0=,&U1=, uint16_t)

//gets called once at boot. Do all initialization that doesn't depend on network here
void userSetup()
{

}

//gets called every time WiFi is (re-)connected. Initialize own network interfaces here
void userConnected()
{
	connectedWiFi = true;
}

//loop. You can use "if (WLED_CONNECTED)" to check for successful connection
void userLoop()
{
	// Send ping request to keep the web server alive
	if (connectedWiFi && millis()-lastPingTime > CheckWiFiDelayMs)
	{
		Ping.ping(WiFi.localIP(), 1);
		lastPingTime = millis();
	}
}
```

Compile using lastest core and WLED version from master branch.

triggerx, 2020-04-22:
I have all but confirmed that this is issue is presented in a Unifi AP based network.  I'm 0 for 5 on NodeMCUs and 0 for 2 D1 Minis on my home Unifi based network when trying to run WLED 0.9.x.  Just keeps resetting.  They work fine on my RAVPower travel router AP.  However EVERY one of those devices work fine on my network when running WLED 0.8.x.  So, I'm almost ready to conclude that *something* in 0.9.x is just not compatible with *something* in my Unifi network.  Also, as an fyi, I have my APs set to 2.4GHz.

AnshumanFauzdar, 2020-04-22:
Hey @gegu Thank you so much for this wonderful idea I will bookmark this one for future updates

@triggerx This maybe the issue you are mentioning, but after flashing my wemos d1 mini with latest firmware and powering both WS2812 an wemos with same power source solved all my problems!
Thanks

Operation760, 2020-04-24:
> I have all but confirmed that this is issue is presented in a Unifi AP based network. I'm 0 for 5 on NodeMCUs and 0 for 2 D1 Minis on my home Unifi based network when trying to run WLED 0.9.x. Just [...]

I tend to agree with this. I have an all-Unifi network. However, my NodeMCUs are not the issue and do not exhibit the problem. It's Arilux ESP8285-based RGB/RGBW Controllers which have WLED firmware which do, When I flash them to Tasmota, the problem goes away.

dpalecek, 2020-04-25:
I can confirm that I have the same exact problems with 2/3 WLEDs appear offline. I am also running a Unifi network

acorn1sh, 2020-05-12:
I am having the sames issues on a LUXUL network. I am using a static IP address but the IP doesn't even show on the router when I search for connected clients. I have tried 4 NODEMCU and all are having the same issue. I think the last version i had that worked was 8.6. I really like the segments so would like to upgrade.

thedannymullen, 2020-05-12:
If you set the static ip on the wled controller it may not show in the router.   Typically the router won’t know about something unless it issued the dhcp address for it.

triggerx, 2020-05-13:
> I am having the sames issues on a LUXUL network. I am using a static IP address but the IP doesn't even show on the router when I search for connected clients. I have tried 4 NODEMCU and all are [...]

Turn off Alexa Emulation in the Sync Interfaces screen.... that'll fix your problem.

acorn1sh, 2020-05-13:
Tried turning off sync services and got one to run for an hour. I started to wonder if it isn’t my APs so I took an old router I had laying around and set it up. I didn’t connect it to the network but set up a SSID to to connect my phone and the modem I nodemcu. I have been stable all day so far. On my main network I have my SSIDs hidden could this cause an issue with the connection dropping? Is it possible that a recent firmware update on the AP is making WLED drop?  I can use the old router but it is a pain as it is not on my network.

gegu, 2020-05-24:
If anyone is interested:

I modified the **unreachable network services ping fix** to keep the code as minimal as possible (external dependency removed) and to make it work with DHCP. [See here](https://github.com/gegu/WLED/tree/master/usermods/Fix_unreachable_webserver)

Aircoookie, 2020-05-24:
@gegu awesome! Would you be interested in submitting the usermod as a pull request? I would then ask people with WiFi issues to give it a try and if it helps a lot of users I would add it to the base code as an option :)

gegu, 2020-05-24:
> 
> 
> @gegu awesome! Would you be interested in submitting the usermod as a pull request? I would then ask people with WiFi issues to give it a try and if it helps a lot of users I would add it to the base code as an option :)

Gladly, it's done.
Since it is not really an extension of the functionality, I have not yet created a PR 😄 
But it's true, the connection problems are unfortunately quite common and affect many users.

It would be great if there was a setting option for it. Of course, only if it helps others as well..

acorn1sh, 2020-05-27:
Would love to give it a try but still new to this. How would I compile for a nodemcu?

gegu, 2020-05-27:
See the wiki section **[Compile settings - ESP8266](https://github.com/Aircoookie/WLED/wiki#compile-settings)**

Aircoookie, 2020-05-27:
What might also be of interest for a few of you is https://github.com/Aircoookie/WLED/issues/946#issuecomment-634100863
That surprisingly helped a lot of users.

acorn1sh, 2020-05-27:
> What might also be of interest for a few of you is [#946 (comment)](https://github.com/Aircoookie/WLED/issues/946#issuecomment-634100863)
> That surprisingly helped a lot of users.

This has fixed my issues so far!

gegu, 2020-05-29:
After some testing time I can tell you that with a WLED version compiled using the latest core version ESP8266 `2.7.1` there are no more WiFi problems in my environment. 
I have disabled the fix and used the default WiFi settings.

gegu, 2020-05-29:
**What I wrote was unfortunately not true.**

It worked because MQTT was active. If I decativate MQTT I need the ping fix again.
So it is clear that outgoing requests help to keep the network services active.

> 
> 
> After some testing time I can tell you that with a WLED version compiled using the latest core version ESP8266 `2.7.1` there are no more WiFi problems in my environment.
> I have disabled the fix and used the default WiFi settings.

closed 2020-10-03 by stale[bot]

Zoltant7, 2020-10-18:
Tried  the latest 10.x Wled,: I got :
wemos d1 mini, with or without leds same,it's looses connection after 5 sec, finally I flashed Tasmota just to see is it a defective unit but seems to work with it just perfectly,

RandomPotato893, 2020-12-01:
I need some help. I've been troubleshooting this for 3 days and I think I've tried everything imaginable. Different boards, different firmware releases, I did the ping thing, I built all the firmware by myself and it seems to not work. Any help appreciated.

AnshumanFauzdar, 2020-12-01:
> I need some help. I've been troubleshooting this for 3 days and I think I've tried everything imaginable. Different boards, different firmware releases, I did the ping thing, I built all the firmware by myself and it seems to not work. Any help appreciated.

Hi @RandomPotato893 
I solved this reboot problem by using same power supply for WS2812 and board (In my case wemos d1 mini)

[32 more replies]