wled/WLED #427: White brightness questions

fijit586, 2019-12-06:
I've attached my layout and rough wiring plan. What I am witnessing is a very definite yellowing of my LED's when I attempt to turn up the brightness. The amount of light increases but the color definitely goes to a yellow hue. At much lower brightness levels the white looks more accurate. I've played with the max amps in the LED setup and it doesn't seem to fix the issue. My guess is I'm running too much current thru my wires. I have injected power at the end of the strips. But these leads are gang wired to the front end leads (thus also in parallel).
If my pdf doesnt come thru, heres a summary of my wiring. I have 4 windows. In each window I have a 5v 30led/m ws2812b strip of approx 120 led's so it's just a cpl feet shy of the full 5m. I have a 5v 20amp PS. I am running my data thru a NodeMCU. The 5v & Gnd powering the strips is wired in para to the MCU. I am currently distributing the 5v & Gnd in para to the 4 strips. I also have a single pixel inline with the data wire to boost the signal to my strips. All FX seem to run fine and all other colors seem to be accurate. My cable is 18awg but my connections to the end of each strip are probably 22awg maybe 24.

I know this is probably an easy answer, but I dont have much electrical experience. I'm just a tinkerer.

[Freds LED setup.pdf](https://github.com/Aircoookie/WLED/files/3932746/Freds.LED.setup.pdf)

brentbrooks70, 2019-12-06:
@fijit586 Your diagram looks correct but your power supply is technically to small to drive that many LEDs at full white, aside from using power limiting. If you have your power limit set to something like 16000 (20A = 20000mA * 80% = 16000mA), as your PS should never run at 100%, at least not for long, and you are still seeing yellowing towards the ends you probably need larger wires to carry the power injection. Do you have a multimeter you can check the voltage at the ends of the  power injection wires? How long is the injection cable from PS to last injection point? 

From [Calculator.net’s voltage drop calculator](https://www.calculator.net/voltage-drop-calculator.html) if I set it to 18awg for 10feet at 5v 30A you would end up with:

>Voltage drop: 3.83
Voltage drop percentage: 76.63%
Voltage at the end: 1.17 

I set your amperage to 30 as that is what you should be using for 480 LEDs (4 windows at 120 each window). 480 * .06A = 28.8Amps at full brightness white

I use standard household wiring solid core 12awg to run 5v up to 15ft, more than that and you can run up to 23 feet with 10awg solid, not stranded, wire. Stranded wire carries up to 28% more voltage drop than solid.

Good luck, hope this helps.

betadeltadelta, 2019-12-06:
While I won't dispute the math above, as it's totally accurate... I think something simpler might be going on here.  For "white" are you selecting white from the present colors under the color chooser in the app, bottom right?  Or are you picking white from the custom color selector right smack in the middle of the large circle?  

Attached is my setup as of last night.  And with regard to voltage drop, I don't believe all LED strips are as sensitive to it.  As example, my setup I'm using upwards of 30' of 18awg wire from my power supply to the right and left most snowflakes, and white is perfectly brilliant white.  However, if you select that pre-select white as I mentioned on the lower right button on the app, I'd call that more "coffee stained teeth white".  I've been snooping in the .ino to find out how or where to change those preset colors.

Forgive the video quality... quickly last night and quick upload, but you can see the brilliant white is just fine.

https://youtu.be/C2tXwoP4T5A

https://youtu.be/ttTtYOt8vPo

tika911, 2019-12-06:
I love the snowflakes.  How did you build the frame to hold the led's?

fijit586, 2019-12-06:
> @fijit586 Your diagram looks correct but your power supply is technically to small to drive that many LEDs at full white, aside from using power limiting. If you have your power limit set to [...]

fijit586, 2019-12-06:
Thanks brent. I've been revisiting my calcs and for bright white I am under powered. I have a 60A PS on order and I'll see what that does. I suspect it will partially fix the issue, with thicker wires taking it the last mile,

betadeltadelta, 2019-12-06:
> I love the snowflakes. How did you build the frame to hold the led's?

5mm plywood from Lowes.  Sketched on CAD, printed, traced and methodically cut them out with a jigsaw.  Painted white and then peel and stick WS2812B 30LEDs/M.  Lot of soldering...

woody2010, 2019-12-14:
> > I love the snowflakes. How did you build the frame to hold the led's?
> 
> 5mm plywood from Lowes. Sketched on CAD, printed, traced and methodically cut them out with a jigsaw. Painted white and then peel and stick WS2812B 30LEDs/M. Lot of soldering...

Do you have a picture of them? I'm very interested

closed 2020-05-11 by stale[bot]