wled/WLED #428: Alexa App will not discover

betadeltadelta, 2019-12-06:
Setting up additional Node's beyond my xmas display.  I cannot get the Alexa app to find my MCU.  What am I doing wrong?  I have the latest build I compiled myself on Tuesday from master with LwIP v2 high bandwidth.  I've read there is some Alexa discovery code bug, but cannot figure out if it's replaced with 2.6.2 or 2.6.4 core.  I'm on the lastest Arduino core, cannot recall the number at the moment.  I've named my Alexa device as LED for simplicity and the box is checked.  I scan and scan and scan, it will not find the device.  Direction?

betadeltadelta, 2019-12-13:
@Aircoookie any idea on this?  I assume it's just as simple as described above, as there are no directions in the wiki.  Any reason you can think of or bug found that would prevent all of my Alexa devices, fire sticks and Echo dot's from discovery?

brentbrooks70, 2019-12-13:
@betadeltadelta First of all are you sure that you have the disable alexa line commented out in wled00.ino? Second, could you post your tesponse to http://_DeviceIP_/json/info? Lastly, can you list your echo devices, type ( echo, echo plus, dot, show, etc.) and generation of each. 
Could you also check your Alexa App -> Devices to be sure there is no other LED listed in lights?

betadeltadelta, 2019-12-14:
@brentbrooks70 first of all thank you for commenting back. Yes I’m positive I did not comment out any features when I last complied my bin. Also attached is the json output just in case it helps identify anything key. So here’s the deal. After 2-3 hours every day for more than a week, no success. I’ve probably scanned for the device more than 75 times.  I haven’t changed anything with the uploaded bin on this particular node in over a week!! But now.... it’s working from about two hours ago. I could not for the life of me tell you why it wouldn’t find it before. It is the only device simply named LED for simplicity of trying to get this bear to work. At any rate, I have several fire sticks, fire stick 4Ks, a 3rd gen Echo dot and of course the Alexa app on my iPhone. Attached is screenshots of info. I’m baffled as to why it all of the sudden has started working. 
[image]
[image]
[image]

closed 2019-12-19 by betadeltadelta