wled/WLED #430: How to create a bin file

slc7777, 2019-12-07:
Ok how do you update to the test version 9
The version i have is 8.6 but when i open it it is a html5 file no files to over write
I downloaded the WLED master which has the newer files but cant work out how to upload to a bin where i can re flash.
I have also tried OTA but wont work

Thanks again for a great program Aircookie

ironosity, 2019-12-08:
building is kind of involved. I'd give steps but I experienced so many issues that I had to work through 1 by 1 that I don't even know. It's basically, follow the wiki using arduino ide, and fix compile problems. Here's a compiled version for esp2866 compatible board (I think d1 mini is). Unzip it, and try flashing it and see how it goes. 
[wled_1912043_ESP8266.bin.zip](https://github.com/Aircoookie/WLED/files/3936640/wled_1912043_ESP8266.bin.zip)

George-Verboven, 2019-12-08:
Hi Ironosity,

Thanks for the ZIP file.
Works great.
Now running 0.9.0-dev

slc7777, 2019-12-08:
Hi Ironosity,
Thanks again for your knowledge and understanding and time for all the newbies doing WLED
Steve

closed 2019-12-09 by slc7777