wled/WLED #431: Wemos D1 Mini - settings don't save, multiple releases

highground88, 2019-12-07:
Firstly great work - I'm obviously having quite a big issue but I hope to be able to use this for our Christmas tree lights soon, as it works great! (temorarily)

I have a Wemos D1 Mini with 4mB flash - and have tried multiple upload methods (esptool, esphome-flasher, Arduino IDE), however none of these methods provide me with a device that the settings persist after a reboot.

I have tried `esptool.py erase_flash` as per #271 , various SPIFFS settings, upload speeds, and also compiled my own 0.8.6 binaries in addition to  WLED_0.8.6_ESP8266.bin, WLED_0.8.6_ESP8266_ledpin4.bin, & wled_0.8.6_ESP8266_lwip1.bin (and also bins from 0.8.5).

I've also tried using in IDE using '1.4 Higher bandwidth' , with no change.

I'm pretty much out of ideas .... is there anything else I can try?

highground88, 2019-12-07:
Just additionally if it helps .... I've migrated through a bunch of different versions, and now the OTA update constantly results in 'Failed updating firmware!'. Tried with fresh 0.8.5 and .6 bins on both versions (upgrade/downgrade). 

Something somewhere on this board has been messed up?

Mariu86, 2019-12-07:
You can compile the last dev version 0.9 to see how it's working must be a more stable version. I tried to compile but I have some library errors .

highground88, 2019-12-08:
Just compiled dev version 0.9, only changes made were to Wifi. Still not holding new setting changes.

Arduino IDE upload options:
[](https://imgur.com/a/8dzrqHl)

Aircoookie, 2019-12-08:
@highground88 Hi and welcome! You are probably using ESP01 or another board with only 1MB flash. "WLED error 2" means the settings could not be written to flash. Try selecting "Generic ESP8266" in the boards menu and select Flash (1MB, FS none) and saving should work!

highground88, 2019-12-08:
Wow.... @Aircoookie .... I'm very sorry..... I thought I'd check the chip so I just opened up the enclosure and I've actually put a random NodeMCU board into this. You are spot on. I should sleep more.

That said - I take it this board won't be any good past 0.9.0 anyway based on your notation re SPIFFS?

Sorry again for everyone that used some brain power trying to work out my problem!!!

highground88, 2019-12-08:
That all said and done - I still cant get this to work even at 1M no SPIFFS.

The 0.9 version goes on, but as soon as any changes are made, I get boot loops.  Using a binary for various ESP8266 etc results in again not retaining data (as I cant control uploads settings perhaps?)

closed 2020-04-15 by stale[bot]

Skyking721, 2020-05-08:
NodeMcu Error 2 again w/ latest Wled release. hmmm

Aircoookie, 2020-05-08:
@Skyking721 what binary/compile settings did you use. Make sure to use the one without `1M` in the name