wled/WLED #432: Preview, HA Integration, and MQTT Cause Random Flashing (0.90-dev)

jrmche, 2019-12-08:
I get random flashes when MQTT or HA integration is enabled. And now when using the new preview feature, the flashing gets extremely bad. When MQTT is disabled, HA integration off, and preview is off, there are no flashes at all. I'm using a 40A 5v power supply with power injected at the beginning, middle and end. There are no problems with brightness and the issue exists even at very low brightness. And I just verified something else, when it's loading different pages on the web interface, I'll get a brief flash. I'm guessing this has something to with some sort of interference? I don't know much about electronics but maybe there's something I need to do to prevent this interference?

Also, love the work that you have done on this. Amazing and fun stuff!

`{"ver":"0.9.0-dev","vid":1912051,"leds":{"count":486,"rgbw":false,"wv":false,"pin":[2],"pwr":0,"maxpwr":38000,"maxseg":10,"seglock":false},"name":"WLED Boxes","udpport":21324,"live":false,"fxcount":87,"palcount":50,"wifi":{"bssid":"08:9E:08:F5:CD:2B","signal":68,"channel":1},"arch":"esp32","core":"v3.2.3-14-gd3e562907","lwip":0,"freeheap":190164,"uptime":2012,"opt":119,"brand":"WLED","product":"DIY light","btype":"dev","mac":"246f2815bf8c"}`

CTLS, 2019-12-10:
having the same issue, however I am not using mqtt, just the home assistant integration.  It seems to be worse on some effects for me.  I am running 500 LED son an esp32.

CTLS, 2019-12-14:
It is still doing this on the beta.  The colortwinkle effect makes it happen the worst for me.

jrmche, 2019-12-14:
Just want to report that so far, the issue has gone away. I did two things and I don't know which fixed it. One, which is what I suspect the problem was, I had a bad LED that took out more than half of the strip (it was LED 136 of 486). I think it was going bad and was causing the flashing. It was replaced. I also updated to 0.9.0-b1. No more flashing for me.

closed 2020-04-19 by stale[bot]