wled/WLED #433: http://4.3.2.1/settings/leds? is unreachable

klrock, 2019-12-08:
I sucessfully flashed 0.8.5 to a  Lolin V3 nodemcu. 

Lolin NodeMCU Features
• Arduino-Like Hardware IO
• Code like Arduino, but interactively in Lua script
• Event-driven API for network applications, which facilitates developers writing code
• Integrates GPIO, PWM, IIC, 1-Wire and ADC all in one board
• 10 GPIO, every GPIO can be PWM, I2C, 1-wire
• 4M Flash Memory
• Built-in WiFi Antenna 

Accessed WLED app, setup Wifi and rebooted.  Accessed renamed WLED1 and tapped the gear symbol to proceed with setup, but got the following response:
http://4.3.2.1/settings/leds? is unreachable
Other menu items are responsive,but cannot setup lights.  However it seems to work fine when accessed by iPhone.

Using PC is preferable for me.

What do you suggest?

Aircoookie, 2019-12-11:
Interesting. You could try accessing the page without the ? at the end of the URL, but that shouldn't cause issues in the first place... I really don't understand why LED settings would be the only thing to not load.

klrock, 2019-12-12:
So I started over by re-flashing and every thing works correctly now. All is good now and I am learning the ropes.

Do I need to have the Updater flashed too or is the OTA performed within WED?

Thanks,
Kerry

raoulteeuwen, 2019-12-12:
> Do I need to have the Updater flashed too or is the OTA performed within WED?

If you now have a working WLED, OTA is included, check under Settings/Security & Updates.

klrock, 2019-12-13:
> 
> 
>         > > 
> >         Do I need to have the Updater flashed too or is the OTA performed within WED?
> > 
> >     > 
>     If you now have a working WLED, OTA is included, check under Settings/Security & Updates.

Thanks.  That's good news.

closed 2020-04-18 by stale[bot]

aliasmore, 2022-04-17:
Hola buenas sabéis por qué no puedo aguardar la clave wifi etc,antes lo ponía se quedaba aguardo ,ahora lo desconecto de la luz y tengo que empezar de nuevo