wled/WLED #434: Enhancements

handsomejackuk, 2019-12-08:
First of all thanks for the great work love the project and keenly following.. this is almost everything i have been looking for..

couple of things i have noted coming from Tasmota... i dont know if any of these are possible but they would be so handy.

display of network password check box in wifi setup. tick box to enable / disable password
also display of password in Mqtt settings in Sync interfaces menu

while we are on there is it also possible to have a mqtt on / off check box as i dont always want my device to be connected to my mqtt server... ( cloud based ) unless i have missed something is there a way of disabling it and eanbling easily  ?

in the main LED control menu where the colors and presets etc are displayed is it possible to have an indication of wifi strength i find this so handy in tasmota to see how good my connection is with my wifi

also an indication of what FX mode i have picked once i have picked it and the applied  FX pallette..

keep up the great work looking forward to 0.9 version

Aircoookie, 2019-12-11:
Thank you for the suggestions! I am very grateful for the feedback!

> display of network password check box in wifi setup. tick box to enable / disable password

Yes, I believe this could be a good idea during initial setup. Of course the password will stay hidden on later page loads for security reasons.

>  mqtt on / off check box

Yep, definitely should add that!

> indication of wifi RSSI strength

Good idea! I believe I should make a little info page in the web UI that displays RSSI, estimated current, uptime, free memory and other important data (could also be used as an output for usermods, e.g. temp sensor reading).

> indication of what FX mode i have picked

is in fact already part of 0.9.0! Will be out soon :)

handsomejackuk, 2019-12-12:
Thanks for the reply I look forward to 0.9 release. 

Great work .. I love it

billchurch, 2020-01-03:
On the RSSI option. For my ESPHome devices, I use the RSSI sensor so it can be logged in Home Assistant, it would be a good option if this RSSI could be a sensor in HA as well, updated every x seconds/minutes whatever... Happy to help as well.

Aircoookie, 2020-01-05:
@billchurch Does ESPHome use a raw RSSI value (negative dB values) or a relative "signal strength" indicator? The latter is already available via the JSON API `info.wifi.signal`. But I could add raw RSSI as well. Then adding that value to the HA sensor component should be possible!

billchurch, 2020-01-05:
Yeah, it's negative dB values: 

[image]

So that would be awesome... This is what one of my ESPHome entities looks like (over MQTT discovery)

[image]

closed 2022-02-21 by blazoncek