wled/WLED #436: Add JSON API for pixel count, color order, E131 settings

computergeek1507, 2019-12-09:
Add JSON API for pixel count, color order, E131 settings. If these settings were available though a API other software could configure them.

Aircoookie, 2019-12-11:
Hi! Yes I believe I will change the way settings work soon (to JSON) so that you could change each value easily from other software!

That said, I don't know what the use case for this would be. Everything that makes sense to change on the fly is accessible via the JSON state API already. Setting led count, color order and E1.31 channels is only something you'd need to do if you change the connected hardware :)

computergeek1507, 2020-04-10:
Has this feature been added and if so is there documentation on it.

closed 2020-08-15 by stale[bot]