wled/WLED #437: support for SK6812WW in WLED_0.8.5_ESP8266.bin

climy, 2019-12-09:
I have a SK6812WW string of 300 that has not worked—not even a blink.  Several smaller WS2812 RGB strings work fine with the same setup as shown.
Test Setup:
- ESP-WROOM-02 development board 
- HTTPUpdater loaded “WLED_0.8.5_ESP8266.bin” 
- Level shifted with SN74AHCT125N
- BTF-LIGHTING RGBW RGBNW Natural White SK6812 (Similar WS2812B) 5m 60leds/Pixels/m from Amazon.

- Crafted and verified a neopixel string adapter/interface board
- Increase the power limit in LED settings to 30000.
- Signal verified by oscilloscope, 5 V data supplied to the Din pin.
- Actually now the smaller 30 and 15 neopixel strips run error free--no pixel flickering or glitches that I saw with the resistor on the Din before.
- WLED oscilloscope output 1Mhz and 333Khz depending on which trace on the 50Mhz scope--although not sure how to read these.

Next tests:
Teensy LC, 5V buffer (pin 17) loaded with example from NeoPixelBus library (which WLED uses): https://github.com/Makuna/NeoPixelBus/blob/master/examples/NeoPixelBrightness/NeoPixelBrightness.inoyour suggested test SK6812 code.

Any comment?
[image]
[image]

raoulteeuwen, 2019-12-09:
What is the spec of your power supply (V & Amps)? Are you sure your 300 LED strip gets enough power to light up?

climy, 2019-12-09:
Thanks for asking.  Has anyone shown they can run SK6812 with WLED bin?  

I use a 30A 5V Computer Products XL250.  It did 6 AMPs draw to a 64 x 64 LED RGB matrix on full white.  But also use a small 4A Adafruit adapter, injecting power on one end of the string.  Even with a voltage drop, should that work?
[DSA00431630.pdf](https://github.com/Aircoookie/WLED/files/3938201/DSA00431630.pdf)

Mariu86, 2019-12-09:
I have 300 sk6812 LEDs working with a 10 amp power supply , I set the amp in wled to 8 amp and it's working without a problem,

climy, 2019-12-09:
Mariu,

Did you compile WLED or upload a .bin file.

 

Thank you for your input.

 

climy

Mariu86, 2019-12-09:
I tried to compile 0.9 version but I get some library error even after I installed all of them I get some duplicate library errors. But I told you if ws2812 it's working the sk6812 must work to.

brentbrooks70, 2019-12-09:
@Mariu86 There is a new library required for 0.9.0, it is ESPAsyncUDP by menodev. It has been added to the getting started in wiki but not the contributors page yet. Duplicate library warnings are no big deal as long as it say they used the esp variants and not the Arduino ones.

Mariu86, 2019-12-09:
I will try tomorrow to see all the errors. Thanks for tips @brentbrooks70, I have a irremote library modified and I must compile myself to use it. Thanks

chiefymuc, 2019-12-14:
I had a lot of issues with SK6812WW with the level shifter. Since I removed the level shifter everything is smooth. 280 leds!

climy, 2019-12-19:
Follow up.  These 300 SK2812 pixels I got on Amazon are DOA!  I assembled a test circuit that allows all the different strand types and count but there is no function on them.
[image]

[image]

closed 2019-12-20 by climy

thewavelength, 2020-03-27:
@climy have you ever got these specific LED strips to work? I have the same and the same problem you have... Wonder if I have to send them back or if I am just missing something.

climy, 2020-03-28:
No.  i returned them for a gift card in exchange from Amazon.  The supplier, who uses Amazon as the material handling interface, did not try to resolve the issue or verify the product was faulty or I blew them out.  I am trying to locate another supply of RGB and white LED's.