wled/WLED #439: Change GPIO pins for use with sonoff basic

jaedeezy, 2019-12-09:
Hello!
The only .bin I've gotten to flash successfully on the sonoff is the WLED_0.8.5_ESP01_1m.bin
I also tried compiling wled00.ino with and without the changes to NpbWrapper.h using arduino ide but can't get it to work(no WLED-AP is broadcasted). Tried various combinations of flash settings to no avail.
 
 How would I go about changing the gpio pins for LEDPIN and button pin for WLED_0.8.5_ESP01_1m.bin?

toto79, 2019-12-09:
did you set flashmode to dout? 

"board_build.flash_mode = dout" isn't used in the default platformio.ini (that was the root couse for me with the same issue "no AP and sonoff basic")
adding this line to the [env:esp01_1m] section ... and voila AP comes up.

maybe lwip instead lwip2 is an idea, but i didn't need this with my sonoff basic (esp8285)

And if you like to use tx instead gpio 2 set "PIXELMETHOD NeoEsp8266Uart1Ws2813Method" to "PIXELMETHOD NeoEsp8266Uart0Ws2813Method" in NpbWrapper.h (but then no log is allowed!)

jaedeezy, 2019-12-09:
Yea flash mode set to "DOUT (compatible)" in arduino ide Tools menu. Are you using platform io to compile and upload? I don't have have any experience with it but I'll give it a shot.

Also, this is on an older sonoff basic r2 (esp8266ex)

toto79, 2019-12-10:
Yes, i use platformio für compile and upload, just rename the file back to ini, but it is from october 2019, there are some differences to the actual one.
[platformio.ini.zip](https://github.com/Aircoookie/WLED/files/3942970/platformio.ini.zip)
Maby this Tutorial can help: https://github.com/toblum/McLighting/wiki/Software-installation

jaedeezy, 2019-12-10:
Thanks @toto79 you couldn't have made that any easier for me! It's working beautifully now, very much appeciated!

blattmann, 2019-12-13:
@jaedeezy @toto79  Could you please explain how you are using WLED on a Sonoff basic?
Sounds very interesting 👍

toto79, 2019-12-14:
I use it with a voltage free / dry contact relay modification to cut the 5v/12V Led Strip power if there is no need.  ESP is powered with 5v directly in case of ws2812 or with a 12v to 5v converter in case of ws2815 strips.

jaedeezy, 2019-12-15:
I have the sonoff inline to cut power to my 5v power supply when off. I have a capacitive touch sensor connected and also plan on adding a DHT22 temperature humidity sensor.

I previously had it setup with a wemos d1 mini, a relay, and 120vac to 5vdc converter to feed the wemos. Seeing as how the sonoff has everything built into such a small package, it made for a much tidier job.

HetBouHe, 2019-12-15:
Have you had any trouble getting the capacitive touch button to work? Mine
does not switch the relay. Is there an additional library that needs to be
installed?

toto79, 2019-12-15:
@HetBouHe : you can/had to define your Button Pin in NpbWrapper.h file. Only uncomment and choose your Pin.

         _Line 8: define BTNPIN 0     //button pin. Needs to have pullup (gpio0 recommended)_

Gpio1 is Tx
Gpio3 is RX

jaedeezy, 2019-12-15:
Also because my capacitive touch button outputs HIGH when touched, I made the following changes:

In wled09_button.ino:
`if (digitalRead(BTNPIN) == HIGH && !buttonPressedBefore`
and
`else if (digitalRead(BTNPIN) == LOW && buttonPressedBefore)`

And in wled05_init.ino:
`pinMode(BTNPIN, INPUT);`

I'm not sure if a pulldown resistor is needed but mine has been working fine without

blattmann, 2019-12-16:
I need to try that :)
Currently I use the Sonoff Basic with Tasmota to turn on / off the AC -> DC transformer for the D1 Mini / LED Strip (I use a 12V to 5V converter for the D1 Mini) via Home Assistant.

closed 2020-05-11 by stale[bot]