wled/WLED #441: ESP wroom-32

vigne, 2019-12-10:
Hi,

I tried several versions of WLED on multiple Wemos Mini32 v1.0.0 boards (the ones recommend here: https://quinled.info/), but the unit is always stuck in a boot loop, with the following message:

```
[12:32:58]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:32:58]flash read err, 1000
[12:32:58]ets_main.c 371 
[12:32:58]ets Jun  8 2016 00:22:57
```

The same file, flashed on a HelTec Wifi-Kit-32 works fine. Also installing ESPhome on the Wemos boards works fine, so it's not looking like hardware issue. Also I tried it on the USB port of my laptop, as well as connected to a dedicated power unit.

The versions I tried are the following:

- WLED_0.8.5_ESP32.bin
- WLED_0.8.6_ESP32.bin
- WLED_0.8.6_ESP32_ledpin4.bin

For flashing I used  `esptool.py erase_flash` and `esptool.py write_flash 0 <filename>` without any obvious errors.

Any help would be appreciated.

Cheers

andree182, 2019-12-10:
you should most likely flash to 0x10000 (see app partition in https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/partition-tables.html#built-in-partition-tables )

jasonblewis, 2019-12-12:
I was facing the same issue and have managed to work out a solution.

When you initially wrote at position 0, you overwrote the Arduino bootloader and flash layout. You need to reflash all that back on and then write to 0x10000

The simplest way to fix it is to use the Arduino IDE to write any demo ino program to the ESP32. Once that is done and working, then go back and rewrite the WLED ESP32 image to 0x10000

eg:

```
esptool.py --port /dev/ttyUSB0 write_flash 0x10000 WLED_0.8.6_ESP32.bin
```

vigne, 2019-12-12:
Thanks, that worked like a charm.

I simply uploaded the "blink" example using Arduino IDE and afterwards flashed with adresse 0x10000 and now everything as it is supposed to be.

Thank a lot!

closed 2020-01-05 by Aircoookie