wled/WLED #442: enhancement

handsomejackuk, 2019-12-10:
OK thanks again for the great software... 

just a thought on esp32 version.. i only just realised that the esp32 has an internal hall effect sensor... tried this code and it works great...

could this be added to the esp32 version...

just to point out i have not got the esp32 version of wled working myself yet, but this would be a great enhancement to start led sequence just hold a magnet near the module...

 `* 
 *  
 * ESP32 Hall Sensor Example
 */
 
 #define LED 2
 
//================================================
//         Setup
//================================================
void setup() {
  Serial.begin(115200);
  pinMode(LED,OUTPUT);
}
 
//================================================
//         Loop
//================================================
void loop() { 
  int sensor = hallRead();  //Reads Hall sensor value
  Serial.print("Sensor Reading:");
  Serial.println(sensor);
 
  digitalWrite(LED,(sensor < 0) ? HIGH : LOW); //On board LED detects presence of magnet
  delay(500);
}`

brentbrooks70, 2019-12-10:
@handsomejackuk This would be a great [UserMod](https://github.com/Aircoookie/WLED/tree/master/usermods). Perhaps you could write it as such and document it. If you do, be sure to put some architecture checking in there so you are trying to read the HES on ESP32s only.

handsomejackuk, 2019-12-11:
> @handsomejackuk This would be a great [UserMod](https://github.com/Aircoookie/WLED/tree/master/usermods). Perhaps you could write it as such and document it. If you do, be sure to put some architecture checking in there so you are trying to read the HES on ESP32s only.

i would love to contribute to this project but my programming skills are not so good... i wouldnt know where to even start... i can load up arduino files and flash them to my boards but thats my limit really. just thought this may be useful to anyone wanting to use a magent to switch on led sequence with esp32...

Aircoookie, 2019-12-11:
@handsomejackuk thank you for the suggestion! Yes, I agree with @brentbrooks70 that this would be very good as an usermod, because not enough users would presumably need it in order to justify adding it to the base code :) I will leave this issue open so that others can find it easily. ESP32 internal temperature sensor is also something that could be supported.

closed 2022-10-21 by Aircoookie