wled/WLED #446: Feature request: Sparkle effect - another way

fribse, 2019-12-11:
I would like an effect where it has a level of light, and then shortly turns up the level or a single LED shortly, sort a sparkling star effect, that would be perfect for a christmas tree light.
So it should not change color like the current sparkle, but just turn up the level for a single LED for a split second.

fishbone-git, 2019-12-11:
Try setting the primary colour as a bright colour of your choice, then set the background (secondary colour) as the same colour but set the brightness slider down to make it a dimmer version of that colour.  That should give the result you want from the sparkle effect

fribse, 2019-12-11:
You are absolutely right, weehee, I didn't think it also saved the brightness in the 1st/2nd color settings.

closed 2019-12-12 by fribse