wled/WLED #447: LoLin NodeMCU v3 - Bootloop after MQTT-configuration

kennerblick, 2019-12-12:
hello from Germany, I am thrilled about your WLED, thank you very much.
I use WLED with LoLin NodeMCU v3 (ESP8266) and WS2812b. 
Since some time I have the problem of the bootloop. I tried different versions of WLED, different power supplies, several LED strips, Arduino IDE and Visual Studio code with Pio, I used several NodeMCU without fixing the problem.
I tried all binaries and compiled them myself with low and high transfer speeds.
After the program transfer, the ESP8266 ran stable until I connected the LED strips and made the sync settings.
Finally the ESP8266 was only sporadically reachable after configuring the MQTT server and ended up in the boot loop.
But I don't want to do without MQTT, so what can I do?
Best regards with respect for your great work.

Andreas

gipnokote, 2019-12-12:
Did you try flashing "wled_0.8.6_ESP8266_lwip1.bin" version? This is what helped me with NodeMCU v3

kennerblick, 2019-12-12:
yes, this bin i have installed also

closed 2020-04-17 by stale[bot]