wled/WLED #452: HASSIO Plugin Ideas

roblandry, 2019-12-13:
It would be great if there were a plugin much like [TasmoAdmin](https://github.com/hassio-addons/addon-tasmoadmin), but for wled. This could allow updating multiple wleds at the same time, as well as centralized control.

r100gs, 2019-12-13:
Yes, that would be great! And than a possibility to control all the leds by sound

pruwait, 2019-12-15:
+

Aircoookie, 2020-01-06:
I am not in a position to add something like this, but I would be happy to collaborate with someone who whould write such a plugin by making changes to WLED that might be needed for such a plugin to work (although there aren't many I suppose)

austwhite, 2020-02-23:
I think Frenck has written a nice integration for HA.  Might be worth a look.

softhack007, 2023-10-02:
Closing, as there was no activity on this enhancement proposal since 3 years.

Meanwhile HA has something for controlling WLED instances, and control by audio is also possible (thanks to a bunch of brilliant developers 😉 )