wled/WLED #453: TypeError

chadsosborn, 2019-12-13:
After upgrading to version 0.9.0-b1, when I set my effect mode I get a message that states "TypeError: Cannot read property 'success' of null".  It does allow me to change the effect, but this message keeps appearing in a red.

[image]

Aircoookie, 2019-12-13:
Thank you for reporting this! You are not the only one who is experiencing this issue so I would like to know a few details to try and get this working for you!
It seems like the ESP is sending an empty response to the state request.

You don't have to answer all these questions, but it would help fix the issue!

- When you select a new effect and the message pops up, does the background of the selected effect change to a lighter gray tone?
- What ESP board are you using?
- Have you installed one of the binaries? If yes, which?
- If you compiled from source, please tell me which Arduino core version and Lwip version you were using? You can find it in the Tools menu and the board manager
- How many LEDs are you using?
- Are you connected to MQTT or Blynk?

Thank you!

raoulteeuwen, 2019-12-13:
I also have this, just now for instance when i tried to change the brightness in the favorites screen.

- When you select a new effect and the message pops up, does the background of the selected effect change to a lighter gray tone?
> so i had this while doing something else
- What ESP board are you using?
> Afaik it's a "NodeMcu v3 CH340", https://www.aliexpress.com/item/32647690484.html?spm=a2g0s.9042311.0.0.528a4c4dMySLIN 
- Have you installed one of the binaries? If yes, which?
> WLED_0.9.0-b1_ESP8266.bin
- How many LEDs are you using?
> 250
- Are you connected to MQTT or Blynk?
> MQTT was still on, so i switched it off since i use (the latest) Home Assistant .

After a reboot, i fiddled around with some settings and got a JSON error. Not sure what i did that threw that error exactly.

I also have a hard time to get the 'Preset cycle' to stick (when i open 2 strips in my browser, switch on preset cycle on 1, switch tab's, and switch back to strip 1, the checkmark disappears and the cycle is stopped).

Btw, love the preview function, nicely done in the UI as well.

tobbensoft, 2019-12-13:
- When you select a new effect and the message pops up, does the background of the selected effect change to a lighter gray tone? 

No

First click, no reaction.   Second click, "Error message"

- What ESP board are you using?

D1 mini

- Have you installed one of the binaries? If yes, which?

No

- If you compiled from source, please tell me which Arduino core version and Lwip version you were using? You can find it in the Tools menu and the board manager

LwIP: v2 Lower Memory
Arduino Core version 2.5.2

- How many LEDs are you using?

265

- Are you connected to MQTT or Blynk?

MQTT for 1-wire temp with Quinled board

tihoangyeudau, 2019-12-14:
i also have this error with 300 leds

Aircoookie, 2019-12-14:
Found the cause, we are running out of memory because a buffer is allocated twice. Working on a fix!

tihoangyeudau, 2019-12-15:
> Found the cause, we are running out of memory because a buffer is allocated twice. Working on a fix!

Also, when I choose led pin is 3, and go to config number led to 300 on the web, wled also have the same error
(d1 mini, 921600, 80mhz, 1M, v2 HB, Easer flash all content, 300 leds)

Skyking721, 2020-10-14:
Hey there... Just received a few Digi-Uno prebuilt and getting this message almost always when changing slider settings in build 2005030   Thinking of reflashing the D1 but will await any replies here. Thanks again for a great piece if code.

blazoncek, 2021-10-03:
Most likely fixed. 😄 Closing.
If the issue persist due to low memory (ESP8266) try reducing LED count and number of segments and/or fewer usermods.
Feel free to reopen if you found another cause.

closed 2021-10-03 by blazoncek