wled/WLED #454: Request: set direction / start for strip or segments

bagobones, 2019-12-13:
Many effects are directional, it would be nice to be able to set the end or flip/ mirror effects.

Especially when using one strip with many segments as you might have one strip go up and down a rail or up and down a roof where the fire effect will just look odd / upside down or you might want to have the effects originate from the peek of the roof and head in opposite directions using segments. 

A Boolean indicating normal or reversed starting end for the strip or segment is all that should be needed config wise.

Aircoookie, 2019-12-13:
Hi, in 0.9.0 there is a checkbox "Reverse direction" in the Segment details. It should do exactly that!

bagobones, 2019-12-13:
Great. I will take a close look at the beta

closed 2019-12-13 by bagobones