wled/WLED #455: SK6812 with 12V

r100gs, 2019-12-13:
Hello guys,

I read that 12v is supported. Does anybody know if its possible tp bye 12V SK6812 stripes?

Best regards,
Stefan

raoulteeuwen, 2019-12-13:
> I read that 12v is supported. Does anybody know if its possible tp bye 12V SK6812 stripes?

Like this: https://www.aliexpress.com/item/32909279999.html?spm=a2g0o.productlist.0.0.2181671ebYh2QA&algo_pvid=de03ada7-f34e-4f21-9637-98b6b3d8407f&algo_expid=de03ada7-f34e-4f21-9637-98b6b3d8407f-8&btsid=2f5c06c9-a62a-4690-9385-2e97dceaae02&ws_ab_test=searchweb0_0,searchweb201602_8,searchweb201603_55 ?

r100gs, 2019-12-13:
Yes, great I think thats it

raoulteeuwen, 2019-12-13:
Also good quality info at https://quinled.info/2019/06/03/what-digital-5v-12v-rgbw-led-strip-to-buy/ ... Please consider closing this ticket so as to limit the open tickets the devs need to pay attention to.

closed 2019-12-13 by Aircoookie