wled/WLED #456: Cannot connect to WiFi network.

klrock, 2019-12-13:
I have entered the Wifi name and password (same as for my iPhone)in WLED-AP several times but after each reboot I get
Client IP: Not connected
My WiFi signal is very strong.
Any ideas?

chunkysteveo, 2019-12-13:
We would need to know what build version you are using, and what board you have uploaded it to @klrock .

Steve

guajabeira, 2019-12-13:
I have a similar issue, on two different NodeMCU's , have flashed and them added to my personal network. I can connect by using the web browser on the laptop, thru 4.3.2.1 or I can also see it if I use the WLED network connection, But it does not show up on the APP and it doesn't show an Ip address either

guajabeira, 2019-12-13:
or and I have an NODEMCU ESP8266mod

guajabeira, 2019-12-13:
using both 85 and 86

chunkysteveo, 2019-12-13:
Sounds like a boot loop @guajabeira. Are you uploading the .bin binary files from @Aircoookie or are you using Arduino and compiling your own from the source?

guajabeira, 2019-12-13:
uploading .bin from @Aircoookie , But I'm a complete rookie, haven't even started with Arduino, yet , lol

Bikemad-uk, 2019-12-13:
I too have similar problems and I'm using an Amica NodeMCU.
WLED_0.8.5_ESP8266.bin works fine, but if I install WLED_0.8.6_ESP8266.bin or WLED_0.9.0-b1_ESP8266 it will not connect to my home network after entering network name and password.

I have tried with DHCP and static IP settings but can only access it via WLED-AP (4.3.2.1) and the device does not appear on my router's DHCP Client List.

WLED-AP wireless keeps dropping out and phone reconnects to home network.

guajabeira, 2019-12-13:
any help. As stated the only way to connect is by picking the WLED-AP network, 
Password to my network appears 2 b correct, static IP (leave at 0.0.0.0 for DHCP)shows
0.0.0.0.
static gateway (shows) 0.0.0.0
static subnet mask( shows) 255.255.255.0
the client IP: (shows)not connected
, AP SSID (shows) WLED-AP , 
with a password filled in
Access Point WIFI Channel ( shows) 1
AP opens no connection after boot
AP IP 4.3.2.1

the Node MCU is an Amica

klrock, 2019-12-13:
I have two nodemcus with the same "not connected" result on two different WLED versions.
Lolin V3  0.8.5
Amica     0.8.6

raoulteeuwen, 2019-12-13:
There are lots of possible reasons your device can't connect. Just rule out:

check what bin-file you used. Did you use the correct file for your controller (if reporting problems, share what controller you have and what bin file you used) 
share how you flashed the file (see for instance https://github.com/Aircoookie/WLED/issues/441 )
your router doesn't filter on MACs or something? Any device is allowed to connect and DHCP is enabled?
your router does (also) offer a 2.4Ghz signal?
the wifi signal is strong enough? How do you know for sure the signal is not the problem?
can you try without the LED strip connected or powered to see whether that makes a difference?
flashing could be a problem: try alternative ways
would be great if you can try another controller (another NodeMCU, Wemos etc), to rule out hw specific issues

klrock, 2019-12-13:
My responses:
check what bin-file you used. Did you use the correct file for your controller (if reporting problems, share what controller you have and what bin file you used)

Lolin V3 0.8.5
Amica WLED_0.8.6_ESP8266

share how you flashed the file (see for instance  #441 https://github.com/Aircoookie/WLED/issues/441  )
NodeMCU-PyFlasher-4.0-x64
Baud rate 115200
DIO
Wipes all data

your router doesn't filter on MACs or something? Any device is allowed to connect and DHCP is enabled?
No MAC filter
Any device allowed
DHCP enabled

your router does (also) offer a 2.4Ghz signal?
Yes 2.4Ghz signal

the wifi signal is strong enough? How do you know for sure the signal is not the problem?
Very strong signal.  Still strong twice as far away.

can you try without the LED strip connected or powered to see whether that makes a difference?
No difference when LED disconnected

Bikemad-uk, 2019-12-13:
Here are the details of my Amica NodeMCU flashed using ESPHome-Flasher-1.2.0-windows-x64.exe:
_Using 'COM7' as serial port.
Connecting....
Detecting chip type... ESP8266
Connecting....

Chip Info:
 - Chip Family: ESP8266
 - Chip Model: ESP8266EX
 - Chip ID: 0079F720
 - MAC Address: 2C:F4:32:79:F7:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)..._

After flashing with WLED_0.8.5_ESP8266.bin this line appears in the WiFi settings: 
[image]
However, the same line does not appear when flashed with WLED_0.8.6_ESP8266 or WLED_0.9.0-b1_ESP8266.
I also noticed that when I reboot after entering my network details, the password only has 8 placeholders visible on versions 0.8.6 and 0.9.0:
[image]
Whereas 0.8.5 (which connects instantly to my network) has all 10 placeholders for my 10 digit wireless password:
[image]
Could it be that the last 2 digits of the WiFi password are somehow not being saved, which is why it is unable to connect to my network via WiFi?
This might explain why it keeps dropping the WLED-AP (4.3.2.1) connection if it keeps trying to connect to the wireless router using an incomplete password.

raoulteeuwen, 2019-12-13:
@klrock could you try the latest 0.9.0 at https://github.com/Aircoookie/WLED/releases ?

You also state "My WiFi signal is very strong." > how do you know your NodeMCU has a very strong signal?

Could you try using esptool to flash?

handsomejackuk, 2019-12-13:
flashing with espeasy flasher.. works on all my other .bin files 0.8.5, 0.8.6 and tasmota,

module is wemos d1 mini reports 4mb memory 

have flashed blank 1mb file before flashing seems the only way i can get it to just about work..

I can connect to AP but very buggy tried all 0.9 bin versions on aircookie wled release 0.9... tried ota update but none working going back to 0.8.6 as seems only stable version for me

Aircoookie, 2019-12-14:
First of all sorry for these issues! I do believe that the issues are due to flash settings somehow.

@klrock @Bikemad-uk could you try the DOUT flashing mode? That is supposedly quite compatible.
@Bikemad-uk That definitely looks like a problem with the wifi password rather than with the connection itself! Do the last three characters of your wifi password contain any special characters?

@everyone a general recommendation from me is using esptool. The other flashers also work 90% and I believe you should use what works best for you, however it seems to result in very hard to debug flash issues at times... esptool comes directly from espressif and i never had any issues with it (although i mostly upload directly from arduino IDE). I will take a minute tomorrow and make a small guide with the commands i'd recommend for easy flashing!

klrock, 2019-12-14:
DOUT Gives same result. No IP.

klrock, 2019-12-14:
I am using NodeMCU-Pyflasher which is a GUI for esptool.py

klrock, 2019-12-14:
0.9.0 does not connect.

I was responding about the router signal.  The NodeMCU sits on top of my desk in the open so I assume it is strong.  The app shows four bars as does my router.

I am using NodeMCU-Pyflasher which is a GUI for esptool.py

highground88, 2019-12-14:
Seems to be the same issue I was experiencing with #431  . I've just replaced the NodeMCU0.9 with a Wemos D1 Mini, and get the same problems as above - both with the .bin and compile.

Bikemad-uk, 2019-12-14:
@Aircoookie I can't see any user adjustable settings in esphome-flasher _(apart from COM port and .BIN file selection)_ so I cannot try the DOUT flashing mode. 

My password only contains letters and numbers, no special characters.

I have 2 Amica NodeMCUs and they flash fine every time and work perfectly with version 0.8.5.

I have also tried a different wireless router, but 0.8.6 and 0.9.0 won't connect to that either.

chunkysteveo, 2019-12-16:
Can people with the issues here confirm they can try uploading a WiFi Example sketch from Arduino to the boards? I.e. go to Examples->ESP8266 and choose a WiFi sketch. Fill in your connection details and open the serial port to confirm the ESP is working and connecting just with a raw "wifi blink" sketch?

Bikemad-uk, 2019-12-17:
ESP8266 Minimal HTTP Updater was initially uploaded using ESPhome-flasher via USB cable.
I then installed WLED 0.9.0 using ESP8266 Minimal HTTP Updater and it uploaded OTA without any problem. 
I then rebooted and accessed it via WLED-AP to enter my network name and password along with a static IP and static gateway details.
After saving and rebooting, it was not detected by my router, or accessible from my network using the static IP I had entered.
I had to access it via WLED-AP and was able to instal 0.8.5 using the OTA update within WLED.
After saving and rebooting, I was then able to access it correctly via my network using the IP address that I had previously entered within 0.9.0. This means that 0.9.0 is definitely  saving the network details correctly, but for some reason is not able to connect using them.
Version 0.8.5 always installs correctly and works fine for me without any problems whatsoever, so I guess it must simply be an incompatibility problem with my [Amica nodeMCU](https://www.aliexpress.com/item/32408097207.html?spm=a2g0s.9042311.0.0.27424c4dCa1BfT) and WLED versions 0.8.6 and 0.9.0, as I can't get any of the later 8266 versions to connect to my home network.

triggerx, 2019-12-19:
So, total N00b to WLED (thanks DrZzzs!), and have been playing around with 8266 NodeMCU's for several months now.... just wanted to throw in my 2 cents.  I'm not smart enough to be able to explain the phenomena, but what I've seen when using devices like this, is that they can't connect because they are not correctly connecting to the 2.4GHz signal.  So, to elaborate... you say your "signal is strong" and your router "also puts out 2.4GHz"... both of which I'm sure I can agree with.  *However* does your 2.4GHz signal share the same SSID as the 5GHz signal?  Basically, if you have just one WiFi SSID, say MYWIFINETWORK, then the NodeMCU is trying to connect to it, but for some reason is trying to connect to the 5GHz signal, but obviously can't.  Again, I can't explain it, perhaps it's the router thinking your NodeMCU is compatible with the 5GHz signal and trying to put it on that.... don't  know.  This is also why it seems to sometimes work, sometimes not, and otherwise is just random.  So, the solution I've found in order to make my "IoT" life much easier is to split the network SSIDs at the router/AP.... for example, 5GHZ is on MYWIFINETWORK5G and 2.4GHz is on MYWIFINETWORK24G.  Same passwords for both, but I just make sure my NodeMCU's connect to the 2.4GHZ SSID in the configuration, and then there are no if's and's or but's about which signal it's connecting to.  Anyway, hope this helps... either with this problem, or your future endeavours.

klrock, 2019-12-19:
triggerx,
Thanks for the suggestion.  It had not occurred to me that the NodeMCU might try to access the 5 Gz network since that had never been an issue with my previous experience with Sonoffs, Tuyas, D1 minis or NodeMCUs, but your reasoning sounded solid.  So I renamed both of the 5 Gz networks and tried again.  The WLED AP app appears fairly quickly in the WiFi list, but I don't see the the NodeMCU on the router after 5 minutes and opening the AP shows that there is no  IP for it.

Thanks anyway.
 Next idea?

klrock, 2019-12-20:
I believe I may have found the solution.  After I changed the name of my two 5 GHz networks, my wife began to complain about no internet.  So marital duties immediately took over the priority for solving.  I had not realized she was using the 5 GHz network and was looking at that when I was reminded once again that an ASUS router update was available.  I had seen it before, but was always in a hurry with no time to wait several minutes for an update.  So I did it and found that suddenly my NodeMCUs were connecting and would reconnect easily.  I do not know the specific issue that caused the problem and am just glad to be moving forward again.

Thanks for all the comments and suggestions.  Looking forward to contributing to this support as well.

Kerry

closed 2019-12-20 by klrock

guajabeira, 2019-12-23:
so 4 different NodeMCU, Amica brand, tried versions 8.5, 8.6, 9.0, changed/upgraded routers, still have same cable company provided modem. use espflasher. over and over. Still can see Client IP, followed all listed above, only thing is haven't tried is another flasher, would that work?. If so how can I get a different flasher. ANY Help

klrock, 2019-12-24:
USE NodeMCU-PyFlasher-4.0-x64.  It is a GUI for esptool.py.  Easy to use and works well for me.
https://github.com/marcelstoer/nodemcu-pyflasher/releases

guajabeira, 2019-12-24:
OK Tried that still no client IP, at first I was flashing without selecting erase flash, so the new data was not showing, I tried both DUAL I/O, then Quad I/O, new nodemcu.

Mariu86, 2019-12-24:
I use the http-ota-update made by Aircoookie and it's working ok to upload the bin file, but now I use the Arduino to compile and it's working ok, I also have the NodeMCU Amica board.

klrock, 2019-12-24:
Also try DOUT.

guajabeira, 2020-01-03:
I tried DOUT still did not work nor did any other options worked. IS there someone that can assist. Currently only able to communicate directly with the NODEMCU, by using WLED network, but the NodeMCU doesn't show an IP address. Was able to program each of the two boards with the WLED Merry Christmas setting, but not able to do any shows or add to xlights as I couldn't find its IP.
What else can I try

raoulteeuwen, 2020-01-03:
> What else can I try

I find it a little hard to find out what you exactly are using and what exactly you tried, so sorry if i ask 'stupid' questions. If i'm right you have 4 Amica NodeMCU's. Does Amica only have 1 type of board? I read at https://frightanic.com/iot/comparison-of-esp8266-nodemcu-development-boards/ they use a CP2101 instead of a CH340. Could the CP2101 cause problems (https://www.google.com/search?q=cp2102+problem)?

I don't see you mention you try to access the NodeMCU on the AP WLED activates on 1st boot. Could it be you missed that part, and the WLED is still in AP-mode? Like mentioned at https://github.com/Aircoookie/WLED/wiki ?

On that same page, Aircookie recommends using esptool to flash, including parameters/config to use.

Can you 1st try to just power the NodeMCU (after flashing attempt) via USB and see whether you can connect to it, before also connecting USB-strips and powering it via a separate power supply?

guajabeira, 2020-01-27:
This us what I have, Just bought new chips Also AMICA. But I still cant get a Client IP. what else.

raoulteeuwen, 2020-01-27:
>  This us what I have, Just bought new chips Also AMICA. But I still cant get a Client IP. what else.  

1. did you try to just set a static IP? What is the result if you try?
2. a couple of times in this thread it is recommended to use esptool as flasher. Can you confirm you've tried that?

guajabeira, 2020-01-27:
I tried the esptool,  same result. When u say static IP, how would that be done. do I just create one? ,  where do I get it from, Im really new af this

raoulteeuwen, 2020-01-27:
> I tried the esptool,  same result. When u say static IP, how would that be done. do I just create one? ,  where do I get it from, Im really new af this

Do you have an Android phone? Can you install the app Fing, https://play.google.com/store/apps/details?id=com.overlook.android.fing&hl=en ? You can use it to find out what devices are on your network. Consequently, you know what addresses are not in use. Home networks should be one of the following

-     192.168.0.0 - 192.168.255.255 (65,536 IP addresses)
-     172.16.0.0 - 172.31.255.255 (1,048,576 IP addresses)
-     10.0.0.0 - 10.255.255.255 (16,777,216 IP addresses)

Say you find your network is using 192.168.1.x, you find your router at 192.168.1.1. Fing shows you no device is using 192.168.1.100 > you can try setting WLED to that IP and see whether you can access it if you give it a fixed IP. But depending on your set up, this is not a permanent solution. I think it is good if you google and read up on home networks.

guajabeira, 2020-01-27:
u r gonna color dumb, tried to 192.168.1.101 , tried with Subnet mask at 255. 255.255.0 , and also fing and it does not show, Is there some way to speak to someone

chunkysteveo, 2020-01-27:
I would recommend you go back to basics and learn how to get a basic blink sketch from Arduino IDE to your boards. From there learn how to get the wifi working with example wifi sketches. Then you will be able to troubleshoot any issues you have with WLED.

guajabeira, 2020-01-27:
ok is there some where specifically to start researching and reading that info,  I do want to thank you for your patience and help,

riemers, 2020-01-30:
Just to add to the list, i had my SSID as "n176" but if you type it, it makes it capital N, so i ended up with N176 and then it doesn't connect. So make extra sure to double check capitals, it is that picky :)

guajabeira, 2020-01-30:
Dude, U my Hero, exactly want it was a freaking Letter Capitalized.Thank YOU So much

chunkysteveo, 2020-01-31:
haha, nice spot @riemers ! I think it goes without saying that SSID's as well as passwords ARE CASE SENSITIVE - curse your phones for auto-capitalising first words, lol!

mberkhof23, 2020-04-13:
I had the same since yesterday when I reflashed my D1 mini with the latest xxxx8266.bin. It turned out that I had fried the D1 by connecting too many leds without proper power supply. (I flashed without the LED power supply connected and changed the configuration to drive too many leds).  The tricky part was that flashing was no problem and even the WLED AP worked but wouldn't connect to my router. When I flashed in the exact same way with a fresh D1, it instantly worked.

guajabeira, 2020-04-13:
My issue was that my network name was not exactly the same as I was typing it. The first letter keep beign type as a capital letter. finally once I paid attention it work and typed correctly  Good Luck