wled/WLED #458: Updated, but NOT GOOD AT ALL

waynerigley, 2019-12-13:
Completed the update OTA onto my Node-MCU and can connect to it no issue vai web, but have absolutely NO control over it what so ever, no On/Off dimming nothing. reverting back to prev version

raoulteeuwen, 2019-12-13:
What version did you run before? Did you already try to cut the power and boot it up again? Also what type NodeMCU exactly do you have? Could it be a 1MB device now being flashed with a file that doesn't fit?

waynerigley, 2019-12-13:
prev vr WLED version 0.8.6 working fine with Hassio via Node-Red updated to WLED version 0.9.0-b1 and broke, confirmed all user settings were correct, i can connect via ip address and get the dashboard but NO control over LED's for any command

raoulteeuwen, 2019-12-13:
Did you do a disconnect from power and put back power?
What board do you have exactly? 
(Can you go back to a working 0.8.6 ? If so, maybe wait for an improved 0.9.0 release?)

Aircoookie, 2019-12-14:
Sorry you are having issues. Did you perhaps use a different LED pin in 0.8.6? Or is the first segment not set correctly somehow? (it should be the entire strip by default)

Patrick-Plueard, 2019-12-14:
While this is a **Pre Release** I think it is going in the right direction But the App Has A long way to go as it keeps crashing on my Phone.
I am having no problems with WiFi as of yet but I am Sure there are a few Bugs Still to be found.

Keep up the good work

waynerigley, 2019-12-18:
Hey guys, it was the same board, same pin ( soldered on ) nothing changed on my side at all, just did the update and NaDa..

closed 2020-04-23 by stale[bot]