wled/WLED #459: Add translation

pepeEL, 2019-12-13:
Hi
Maybe a good feature request will be add multilanguage ? I can support translation to POLISH language. What do you think ?

Aircoookie, 2019-12-14:
Hi, yes I do believe that is a very good idea! I still have to add the necessary infrastructure to support multiple languages though. Also I already have other users who would be doing translations to Spanish, Russian, Danish, maybe Japanese and German (myself haha). So I will think about adding that language support soon, at least for the main UI!

pepeEL, 2019-12-14:
You can inform US on this thread when you add?

Aircoookie, 2019-12-14:
Sure!

keesg60, 2019-12-18:
Please add Dutch also when you’re ready. I’m willing to do the translation.

Vvasyash, 2019-12-28:
I’m ready for Russian 

> Hi, yes I do believe that is a very good idea! I still have to add the necessary infrastructure to support multiple languages though. Also I already have other users who would be doing translations [...]

deusyv, 2019-12-29:
and me for French

Nikolay16700011, 2020-12-23:
Good afternoon, is the theme still alive? Today I picked through the code. It was not difficult to translate the configuration pages, they are open in the IDE sketch. But how to translate the main screen, with control buttons. In the Arduino IDE, it is somehow encoded. The data folder contains html files, but changing their language does not affect anything. How to translate the program desktop correctly.

Aircoookie, 2020-12-23:
@Nikolay16700011 you'd need to edit the files in the data folder indeed, and after that run an npm command to encode the data for the sketch. No need to change anything in the .h files directly: https://github.com/Aircoookie/WLED/wiki/Add-own-functionality#changing-web-ui 

Note that I'll still need to add a custom JSON infrastructure to support dynamic language switching. So you can translate it for your own use, but you won't be able to switch between it and the english version easily.

pepsonEL, 2020-12-23:
But when you plan add translation support ? I also can help with translation POLISH.

Nikolay16700011, 2020-12-24:
Thanks to Aircoookie, it worked!!! There was a problem with the settings page (settings_dmx.html), it had to be corrected manually. When translating the names of buttons other than English, symbols were displayed instead of letters on the buttons. At the top of the page code, I added the line < html><head> <META charset=utf-8>. After that, everything was displayed normally. Recommendations for those who want to make this journey. Make all changes on one page, and immediately try to upload to ESP. To see the error in time. Then proceed to the next one. The names of the effects themselves need to be translated in the Arduino sketch, FX tab.h the names of the effects :const char JSON_mode_names[] The names of the color palettes: JSON_palette_names const char[].

Nikolay16700011, 2020-12-24:
Thanks to Aircoookie, it worked!!! There was a problem with the settings page (settings.html), it had to be corrected manually. When translating the names of buttons other than English, symbols were displayed instead of letters on the buttons. At the top of the page code, I added the line < html><head> <META charset=utf-8>. After that, everything was displayed normally. Recommendations for those who want to make this journey. Make all changes on one page, and immediately try to upload to ESP. To see the error in time. Then proceed to the next one. The names of the effects themselves need to be translated in the Arduino sketch, FX tab.h the names of the effects :const char JSON_mode_names[] The names of the color palettes: JSON_palette_names const char[].

DarkAssassinUA, 2024-10-06:
Still waiting possibility to translate

Sojourneer, 2024-10-13:
> Still waiting possibility to translate

I'm well under way as described in #3180. Google Translate gives a quick translation when ch can be tweaked.

Sojourneer, 2024-10-19:
This is what I am working on.  From testing, I believe that both International and Translation (thanks to googletrans API) can be done very quickly.

# Localization Proposal

## Client scope
1. Display language only, including text delivered to browser via JSON api.  Date formats, etc. are not covered.
2. Display language is a property of the WLED instance.  Each browser client will use the same language.

## Translation features
1. Translation comprises an Internationalization step and a separate step of Translation for each language
2. Both Internationalization and Translation are incremental process, which allows a process of test-and-amend, as well as reuse of previous work on new builds.  This is facilitated by using a hash to identify each piece of internationalized text
3. The Internationalization process replaces target text in HTML files with placeholders referencing a hash of the original text.  The result is called a Templatized file.  Script that includes target text moves it to the including HTML in a hidden section.    

## Runtime structure
1. The display language is set in the xxxx settings page
2. WLED serves templatized files. The first script to run in any HTML page replaces the template's placeholders with translated text obtained from the translation dictionary.
3. The translation dictionary for the current display language is retrieved from a central repository (e.g. github), and cached in Window:localStorage. 
   1. Refetching can be forced in the xxx settings page. 
   2. This implies that github must be available to each browser client device the first time it accesses WLED, whenever the display language is set, or when refetching is requested.

DarkAssassinUA, 2024-10-29:
> This is what I am working on. From testing, I believe that both Internationalization and Translation (thanks to googletrans API) can be done very quickly.

Any progression?

Sojourneer, 2024-10-31:
> > This is what I am working on. From testing, I believe that both Internationalization and Translation (thanks to googletrans API) can be done very quickly.
> 
> Any progression?

Yes.  I have the UI 95% working as static content on a python HTTPServer running in Ubuntu (showToast content is in progress) from the source tree.  My next issue is to figure out where and how to store the localization.js script and the language json files, put the new endpoints in the WLED server, and then to test on an actual ESP.

Sojourneer, 2024-11-04:
> > This is what I am working on. From testing, I believe that both Internationalization and Translation (thanks to googletrans API) can be done very quickly.
> 
> Any progression?

@blazoncek @Aircoookie 
I've had my first test switching languages on my ESP8266 controller, which was successful.  The JSON language files are hot linked from Github without problem.  I noticed one issue on switching to French: the words are too long for the top menu in the Mobile mode.

A little bit of tweaking, and I should be ready to push in a week.