wled/WLED #460: Lost WiFi after OTA flash

fribse, 2019-12-14:
Board: Wemos MINI D1 ESP32
First impression was that it was offline, so I pulled it from the diguno board, and flashed it locally, still offline, until I found out that the AP was active.
By that time I had erased all my presets :-(

Other than that, swipe left/right = bad, if you try to drag the sliders, you will swipe instead.
Preset overview = great, you can clearly see which slots have presets, and the setting of the cycle mode is much better this way.
NTP server setting = great, works perfectly.
Effects preview = great! Makes it so much easier
Preset cycle = bad ( I think), it looks like I loose the preset cycle setting all the time, so that it is stuck on one effect.

I currently only have d1 mini 32 as boards for my wled, but that might very well change when I get more lights up (only have 3 wled's running as of yet), but I can't see in the GUI which file I've loaded in, and thus it makes it cumbersome to update them after a while.

fribse, 2019-12-14:
After flashing the second I can now see that all settings are gone when doing the OTA.
It doesn't really matter for the presets, but the number of LED's makes it a bit less than ideal.
I will of course start noting down all this, which I haven't done yet, but setting the number of LED's in a string is quite annoying to do that long after mounting it, if it's behind a diffuser :-o

Aircoookie, 2019-12-14:
Hi!
Sorry that you lost all flash contents, that's a bummer. I am still investigating why that even happens, I believe it is due to flash sectors not lining up correctly...

The sliders have a property that should prevent the UI swiping when you drag them. What browser/device are you using? Need to look into that!

You can see the installed version ID in the bottom of security settings, that may help out finding which board is which :)

But I am happy that you like the rest of the update and I hope you will have a lot of fun with WLED despite these issues!

fribse, 2019-12-22:
Hi Aircookie
Yes, very very happy, you are doing a great job, and I'm not posting to complain in any way, just want to inform you.
It does say the version, but not the special build for my dig uno boards 😉 

The swiping is on my mobile phone screen (6 something inch display) in Chrome.

closed 2020-02-22 by fribse