wled/WLED #461: ESP01 module bootloop after setting wifi datails

Amli2011, 2019-12-14:
Hi!
I use a ESP01 module 1MB.
I have tried different builds:

WLED_0.9.0-b1_ESP8266_1M_full.bin
WLED_0.8.6_ESP01_1m_full.bin
WLED_0.8.6_ESP01_512k.bin

Transfered with esptool, result:

---------------------------------------------------
esptool.py --port /dev/ttyUSB0 --baud 460800 write_flash --flash_size=detect 0 ../Nedlastinger/WLED_0.9.0-b1_ESP8266_1M_full.bin 
esptool.py v2.5.1
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 80:7d:3a:6e:c7:4c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 511216 bytes to 351550...
Wrote 511216 bytes (351550 compressed) at 0x00000000 in 30.7 seconds (effective 133.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
---------------------------------------------

I can then connect to the wled ap. I can controll leds, configure numbers of leds etc. and al seems to work when run from the AP.

When I set the ssid and password for my network it is gooing into a bootloop.
 (ssid is only letters, first letter is captal - 6 characters, password is a mix of upper and lowercase, a number inside and last character is uppercase, no spesialcaracters - 12 characters)

Here is the output from serial console when bootlooping:

---------------------------------------------

Exception (28):
epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd30 end: 3fffffc0 offset: 01a0
3ffffed0:  00000000 00000001 3fff1cc0 3fff2120  
3ffffee0:  3fffff40 3fffff4c 00000000 4021f26f  
3ffffef0:  00000050 00000005 3fffff20 3fff1ef8  
3fffff00:  00000003 3fff2100 3fff1684 3fffff4c  
3fffff10:  3fffff40 00000050 3fff2120 4021f2da  
3fffff20:  3ffeaa8b 3fffff40 3ffeaa87 3fff1ef8  
3fffff30:  3ffeaa8b 3ffeaa87 3fff2120 40215fae  
3fffff40:  00706374 0b02a8c0 83c6a7f0 64656c77  
3fffff50:  3fff5200 8421da48 401005f8 3fff1ef8  
3fffff60:  3fff1efd 3fff1ef7 00000000 40216216  
3fffff70:  00000000 4021da48 3fff22f8 3fff2338  
3fffff80:  00000000 00000000 00000001 3fff2338  
3fffff90:  3fffdad0 00000000 3fff22f8 4021be71  
3fffffa0:  feefeffe feefeffe feefeffe 4022bb8c  
3fffffb0:  feefeffe feefeffe 3ffe8818 40100465  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbc204a9b
~ld
Ada

-----------------------------

it is repeating. If leds are connected, they will fade in and stay on for some seconds before going straight to black and fade in again. 4 seconds between the cycles. the text above appears right before it goes black. No other output to the serial.

Before I was setting the wifi credentials I did set number of led and changed type to RGB. The setting with 50 leds and RGB type is saved and active as the whole led test strip is in the right color even when it bootloops.

I have tried with diferent ESP01 modules I have, and get the same result. 
My network AP is in free sight 5 meter from my esp. No problem with other software on the esp(TASMOTA is tested)

I am new to the WLED firmware and tried 8.6 last week, same result as. Tried 0.9.b1 now and serial and programming outputs is from this try. I have not tried earlier versions.

To get out of the bootloop I have to erase the flash before I program it again.

(I did try wled_0.8.6_ESP8266_lwip1.bin, and it did connect to my wifi, but did not save, it works as long as power is not interupted. Reason is it is for 4MB and not 1MB. at the moment using that build on an esp12 module and it is working)

Sorry for the long text. But i hope I get all the important info in at once.

Amli2011, 2019-12-14:
Have now tried with DOUT:
esptool.py --port /dev/ttyUSB0 --baud 460800 write_flash --flash_mode dout --flash_size=detect 0 ../Nedlastinger/WLED_0.9.0-b1_ESP8266_1M_full.bin

Still same problem.

I have also tried to set a static ip address (ip address, gateway and mask) - same problem.

raoulteeuwen, 2019-12-14:
> Hi!
> I use a ESP01 module 1MB.
> I have tried different builds:
> 
> WLED_0.9.0-b1_ESP8266_1M_full.bin
> WLED_0.8.6_ESP01_1m_full.bin
> WLED_0.8.6_ESP01_512k.bin

Could you try 0.8.5.?

Amli2011, 2019-12-14:
Tried 0.8.5
Erased flash first, then loaded 0.8.5 for esp01 1MB with DOUT. Works ok, saves wifi credetials and connects fine.

flashed 0.9.0b1 without erasing in hope stored credentials would work, but same bootloop.
then erased flash, flashed 0.9.0b1 entered wifi credentials, verified same bootloop.
Flashed 0.8.5 without erasing and it connected with the credentials stored via 0.9.0b1

FloatingBoater, 2019-12-15:
Hi,
I've seen the same issue on 8.6.0, but didn't have a serial console to capture debug info.

# ESP01 with a 5v regulator
```
esptool.py --port /dev/ttyUSB1  erase_flash
esptool.py --port /dev/ttyUSB1 write_flash -fm dout 0x00000 WLED_0.8.6_ESP01_1m_full.bin
```
# ESP01 Results
* WLED_0.8.6_ESP01_1m_full.bin - boots, can save LED settings without touching WLAN, after setting WLAN password crashes repeatedly (ESP01S blue LED blinks, then LED string flickers) connecting to WLAN, defaults back to own AP.
* WLED_0.8.6_ESP01_512k.bin - same as 1m_full
* WLED_0.8.5_ESP01_1m.bin - works properly (different GUI).

# ESP8285 Method
Wemos D1 Mini clone from Amazon with an ESP8285
```
esptool.py --port /dev/ttyUSB1 --baud 115200 erase_flash
esptool.py --port /dev/ttyUSB1 write_flash -fs 1MB -fm dout 0x0 WLED_0.8.6_ESP8266.bin
```
# ESP8285 Results
* WLED_0.8.6_ESP8266.bin - Can't save settings
* WLED_0.8.6_ESP01_1m_full.bin - boot loop, flashing LEDs.
* WLED_0.8.5_ESP8266.bin - Can't save settings
* WLED_0.8.5_ESP01_1m_full.bin - works fine.

Amli2011, 2019-12-15:
@FloatingBoater 
With the WLED_0.*_ESP8266.bin save should not work, as it is built for a 4MB esp it saves the settings in an not existing part of memory on 1MB devices (8685 and ESP01).

ESP01S (should be black with 1MB)
ESP01 (should be blue with 512kB)
A blue esp01 with the 1MB build should in theory in my head have trouble save settings same way as when using a 4MB build on a 1MB device, and thus reverting to AP as nothing realy was saved.

When I use a 1MB build like WLED_0.8.6_ESP01_1m_full.bin on a 1MB black ESP01 device I get bootloop, blinking and no way to return to AP mode unless I Erase the whole device and program it over again.

Could you confirm the your blue card is 1MB and actualy returning to AP?

to clarify, I have only tested with 1MB black ESP01S units.

SittingDuc, 2019-12-16:
I will add a "me too".
* ESP01S with 1MB flash
* 0.9.0-b1
* using PlatformIO to flash
* configured Wifi and count of LEDs all fine. Moved the ESP01S off the programmer and onto the LED string, device blinks the onboard LED every 10-12s. DHCP server shows it requesting a new IP every 10-12s. Serial console shows

```
wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
Ada
```

raoulteeuwen, 2019-12-16:
> * configured Wifi and count of LEDs all fine. Moved the ESP01S off the programmer and onto the LED string, device blinks the onboard LED every 10-12s. DHCP server shows it requesting a new IP every 10-12s. Serial console shows

So the difference(s) is(are): how the ESP is powered, to what it is connected and the location of the ESP relative to the AP?

SittingDuc, 2019-12-16:
I moved it less than a metre, so position relative to the AP is not that different. But the other two variables are fair.
Actually, I think it continued to boot loop after I moved it back to the programmer. Trying 0.8.6 from the tarball now

SittingDuc, 2019-12-16:
.. because that was an abjectly terrible bug report, time for a better one.
1. the original ESP-01S reflashed to 0.8.6 works Just Fine.
2. grabbed a new ESP-01S and flashed it with 0.9.0-b1 from github.
Disabled OTA Alexa, Blynk, Cronixie, HueSync, Infrared, Filesystem. 
Enabled MQTT, AdaLight, Debug.
Set the SSID, Password and NTP server in wled00.ino; because this thing is going onto an isolated LAN and I can never be bothered with using my phone and switching AP and...

Without power cycling at all, without connecting to the web at all, it says
```
---DEBUG INFO---
Runtime: 261359
Unix time: 1576486252
Free heap: 27896
Wifi state: 3
State time: 101295
NTP last sync: 96120
Client IP: 192.168.xx.xx
Loops/sec: 10985
```
And repeats every... about 10s. DHCP server shows it is not requesting new addresses.

So, on the programmer with zero power cycles and never having touched the webui, the ESP-01S is happy. Time to break it. For Science.

Power cycled to set GPIO0 floating
```
---DEBUG INFO---
Runtime: 70018
Unix time: 1576486465
Free heap: 27928
Wifi state: 3
State time: 0
NTP last sync: 4859
Client IP: 192.168.xx.xx
Loops/sec: 10821
```

used webui, hit the wifi config
```
---DEBUG INFO---
Runtime: 120033
Unix time: 1576486515
Free heap: 27736
Wifi state: 3
State time: 0
NTP last sync: 4859
Client IP: 192.168.xx.xx
Loops/sec: 10774
Forcing reconnect.
Access point disabled.
Connecting to APNAME...
---DEBUG INFO---
Runtime: 130036
Unix time: 1576486525
Free heap: 22288
Wifi state: 6
State time: 130036
NTP last sync: 4859
Client IP: 0.0.0.0
Loops/sec: 10512

Connected! IP address: 192.168.xx.xx
Init STA interfaces
mDNS started
---DEBUG INFO---
Runtime: 140039
Unix time: 1576486535
Free heap: 28240
Wifi state: 3
State time: 140039
NTP last sync: 4859
Client IP: 192.168.xx.xx
Loops/sec: 9750
```

Power cycled again, still good.

Changed the LED count to 29, set the mDNS field to blank to disable it.
Attempted to disable Alexa emulation, it won't let me, because the hue fields on the same page are blank and that's not allowed. Saved those and it still seems good

So, round one goes to the device. With all the features disabled to shrink the image size (mostly) and Debug enabled, no sign of the boot loop while on the usb programmer dongle.
```
DATA:    [=====     ]  51.5% (used 42172 bytes from 81920 bytes)
PROGRAM: [====      ]  45.0% (used 460768 bytes from 1023984 bytes)
```

FloatingBoater, 2019-12-16:
> @FloatingBoater
> With the WLED_0.*_ESP8266.bin save should not work, as it is built for a 4MB esp
Makes sense - the address space exists in the uP, but isn't connected to any storage.

> ESP01S (should be black with 1MB)
> ESP01 (should be blue with 512kB)

> Could you confirm the your blue card is 1MB and actualy returning to AP?
My ESP01 devices are marked ESP-01S S Series with a black PCB material. Next to the ESP8266EX uP, is what I believe is the SPI flash, marked P25QB80H 881F88D but I can't find a data sheet to confirm the flash size.

> to clarify, I have only tested with 1MB black ESP01S units.
Same here.

SittingDuc, 2019-12-16:
Round two. Enabled everything. OTA, Alexa, Blynk, Cronixie, HueSync, Infrared, MQTT, AdaLight, Debug. Disabled Filesystem.
```
DATA:    [======    ]  55.5% (used 45480 bytes from 81920 bytes)
PROGRAM: [=====     ]  48.5% (used 496180 bytes from 1023984 bytes)
```
Loops per second is down around 8200. Free Heap is 23200. Using the webui, some of the settings still seem set; suggests platformio isn't doing a very complete erase/ any erase at all. Not ideal.
Factory Reset. started fiddling with settings to see what I can change.
Disabled mDNS, disabled alexa. set LED count to 29. set OTA passphrase. Power cycled on the programmer and **boot loop**

```
Connected! IP address: 192.168.xx.xx
Init STA interfaces
send NTP
NTP recv, l=48
Unix time = 1576487676

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

---WLED 0.9.0-b1 1912131 INIT---
esp8266 2_4_2
heap 31272
LEDs inited. heap usage ~184
Load EEPROM
Ada
Access point disabled.
Connecting to APNAME...

Connected! IP address: 192.168.xx.xx
Init STA interfaces
send NTP
NTP recv, l=48
Unix time = 1576487690

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

---WLED 0.9.0-b1 1912131 INIT---
esp8266 2_4_2
heap 31272
LEDs inited. heap usage ~184
Load EEPROM
Ada
Access point disabled.
Connecting to APNAME...

Connected! IP address: 192.168.xx.xx
Init STA interfaces
send NTP
NTP recv, l=48
Unix time = 1576487703

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)
```

So, some combination of setting values on my ESP-01S caused it to bootloop. Yay

I will use esptool to erase the flash and try to program again, see if I can work out which setting is The One

SittingDuc, 2019-12-16:
used esptool to erase the flash and reprogram. Erase is required, only reprogram can leave the boot loop functioning.

It seems to require two or three of
* Wifi setup: set mDNS field to blank
* LED Preferences: set LED count to 29
* Sync Interfaces: Turn off Alexa
* Sync Interfaces: disable ON/OFF button.

Noting that reprogram without erase leaves the device in the failed state means, I hacked the wled00.ino and added Serial.println() to try to see where it went wrong.
It seems to get into the main loop Just Fine, then manages to connect to the AP, update NTP, and then goes silent. So it's not as simple as an array-bounds check on reading the configuration settings at power on.

lubeda, 2019-12-16:
i am also trapped in the bootloop. With different D1 Mini Clones.
The .8.6 did not work also 0.8.5 no problems.

dorinik, 2019-12-18:
I used ESP-07 (also 1M), but they are the same errors as with ESP-01.
The latest WLED version that works correctly is 0.8.5.
I mention that only 1M WLED versions have been tried (**** ESP01_1m.bin)

Aircoookie, 2020-01-04:
Thank you for the very elaborate testing! Seems like it bootloops after recieving the NTP time... Does the bootloop still happen if you disable NTP in Time settings?

lubeda, 2020-01-09:
Sorry, i couldn't test it **in detail**, but isn't the ntp setting disabled as standard?

By enabling it i made the 0.8.5 work.

Amli2011, 2020-01-10:
Version 0.9.0-b1, after erasing and then flashing the default is NTP disabled. No difference if I enable it. 
I did also try it on a wifinetwork where it could not reach internet, no difference.

juppwerner, 2020-01-22:
Hi,
I have a wemos d1 mini light and flashed the 1M bin. 
I had the same reboot issue. Using the initial AP mode I configured WiFi and the Leds (type, count) and saved the settings. The rebooting occurs. 

After re-flashing and disabling mDNS it works fine! 

Re-enabling the mDNS again results in boot looping. 

So maybe one should disable the mDNS / leave blank as default for the 1M bin build? 

Thanks for this great project!!! 1*

Regards
Joachim

Amli2011, 2020-02-02:
I can confirm what juppwerner has discovered. When I clear the mDNS field I can make it to connect to my network without bootlooping.

FoxExe, 2020-02-05:
Same here. ESP-01s (1MB).
Disabling mDNS solve problem, but only for reboot.
WLED v8.5.6 works w/o any problems.

If add some debug messages into code, its will stuck on NTP configuration (Receive NTP data), so, seems like its stuck somewhere in interrupts...

SittingDuc, 2020-02-25:
Updated to 0.9.1 and so far no problems; clean git clone and setting similar options in the config and touch wood it hasn't bootlooped yet.

FoxExe, 2020-02-29:
Confirm. All works good on 0.9.1

Amli2011, 2020-02-29:
With WLED_0.9.1_ESP8266_1M_full.bin
first erased the chip, then programmed. working ok in access piont mode. but bootloops after i set the credentials for my wifi.
i still get the same bootloop.

If i set the mDNS blank (disabled) it does not bootloop.
0.8.5 versions are ok. 0.8.6, 0.9.0 and 0.9.1 has the same problem, but only in the precompiled binary for esp01 1MB and 512k.
I have not tried to compile myself. But have tried the other binaries for 4MB without problem.

Serial output when bootlooping:

Exception (28):
epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd30 end: 3fffffc0 offset: 01a0
3ffffed0:  00000000 00000001 3fff1f34 3fff2398  
3ffffee0:  3fffff40 3fffff4c 00000000 402228e3  
3ffffef0:  00000050 00000005 3fffff20 3fff2170  
3fffff00:  00000003 3fff2378 3fff184c 3fffff4c  
3fffff10:  3fffff40 00000050 3fff2398 4022294e  
3fffff20:  3ffeabd3 3fffff40 3ffeabcf 3fff2170  
3fffff30:  3ffeabd3 3ffeabcf 3fff2398 4021c07e  
3fffff40:  00706374 9502a8c0 83c6a7f0 64656c77  
3fffff50:  3fff5500 842210bc 40100b30 3fff2170  
3fffff60:  3fff2175 3fff216f 00000000 4021c2de  
3fffff70:  00000000 402210bc 3fff2598 3fff25d8  
3fffff80:  00000000 00000000 00000001 3fff25d8  
3fffff90:  3fffdad0 00000000 3fff2598 4021f3e7  
3fffffa0:  feefeffe feefeffe feefeffe 402304c4  
3fffffb0:  feefeffe feefeffe 3ffe8800 4010089d  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbc204a9b
~ld
Ada

smith844, 2020-04-15:
I appear to have the same problems and I have tried several solutions but to no avail.  Just wondering what to try next.
I have ESP01S 1M which has had:
WLED_0.9.1_ESP8266_1M_full.bin
WLED_0.9.0-b1_ESP8266_1M_full.bin
WLED_0.8.6_ESP01_1m_full.bin
WLED_0.8.5_ESP01_1m.bin
it previously worked on 0.8.5 but after upgrade to 0.9.1 it boot looped, I then blanked out the mDNS and it worked for a couple of days then failed.
Now it boot loops regardless of what bin file I have used.  I have erased flash between loads.
I am using ESPHome_Flasher_1.2.0 on windows to flash but to be on safe side i have also erased flash using esptools on my mac.
I have tried several ESP01 modules, some which were WLED but some which werent and the effect appears same, regardless of what 'options' i deselect.
result is WLED_AP loads fine, I can change settings in number of LED etc but once wifi settings are put in it then enters a boot loop.
It does connect as I can sometimes, depending on timing, connect and change some options.
I have tried, with no success:
NTP on or off.
static IP or not.
mDNS blank and complete
Alexa selected and not.
Can anyone shed any further light on a fix. 
I have just ordered some 4M chips to replace the stock ones, will this fix the problem ?

ClintH, 2020-04-15:
I've had nothing but reboots with an ESP-01 with every WLED firmware I can find. The wifi AP shows up for some period, but usually when I try to connect my phone it will reboot. I don't even get to access the config page. Tried with several different modules. They all work fine with Tasmota.

Waiting on a second USB adapter I can use to get stack traces, I've only got one that's set up for flashing firmware.

smith844, 2020-04-15:
Interesting.  I have previously been using an MQTT based arduino sketch originally by BRUH Home Automations which does 90% of what I needed.  I modified it a little so it is really good with Home Assistant and you can select usual stuff by mqtt message, effect, brightness etc.  But the rest of it is hard coded in sketch.  I was looking for a little more flexibility which WLED brings, if you can get it to be stable enough.  I think the EPS01S is such a great little module, i can use D1 Mini but i like the physical size of ESP01S.
That said I just tried WLED_0.9.1_ESP8266 on a Wemos D1 Mini and it was stable until i changed to static IP then it got in boot loop as well.  I am going to get stack dump and publish on here, later.
It will be a shame if I cant use it as WLED is a fantastic piece of work, and credit to @Aircoookie

Amli2011, 2020-04-16:
@ClintH your symptoms are not the same as this thread, as this thread is about bootloop after setting wifi details and reboot. Are you sure you have a good powersupply (not all breadboards are 100%, and bad connections happends)? No errors during writing of the module? Using the 1M versions of the bin file? Have you tried to contact the wled ap with a computer or only by phone? I have seen some dnsproblem when using phone (the phone dont find internet on wled accespoint, and then tries to use mobile internet to reach the wled, only to not get contact, and the wled times out....) I don't know if any of this is relevant, but it is problems i have had.

Amli2011, 2020-04-16:
@smith844 Your first problem seems to bee the same problem I have. I have used a ESP01S modified with a 4MB chip, and it works with the ordinary binary files (have not tried the 1M files with the 4MB chip)
When I follow the tip about disabling (leave blank) mDNS, I do not have any problem. It is then stable, and have been for months.

I have had some hardware problems with my circuit using the ESP01S because the two exposed GPIO have to be high on boot, and my pullup resistors was not good enough, resulting in the module going to some program load modus on startup. This should have been the same issue with 0.8.5.
After you got the more permanent problem, have you tried to go back to 0.8.5 in case there are external sources of the problem?

antivirus68, 2020-04-16:
I'm having the same problem and error log with this NodeMCU: AZDelivery ESP8266 NodeMCU Amica V2 Modulo ESP-12E WiFi Development Board con CP2102

I flash the WLED_*_ESP8266.bin version and I followed the DrZzz tutorial to make it work with a led strip.

The weird thing is that the first time I installed 0.9.1 everything worked fine. No bootloop at all with mDNS configured. I connected to a led strip and everything was fine.
Today for no reason, it started to bootloop. The only thing I did was to turn off the power supply and on again because I had to move the led strip in another room.
I've tried re-flash to 0.8.5 and 0.8.6. I've tried to disable mDNS. No luck.
It works only on AP mode.

I've a Netgear mesh wi-fi with only one ssdi for 2.4 & 5 so I guess that, since NodeMCU wifi works only with 2.4Gh it connects automatically to 2.4Gh.  

I ping the device while it bootloops and for few seconds it is connected to the network since it responds.

I don't know what to do more. And most of all I'm worried that if I buy another one it will have the same problem.

Any suggestions?

Thanks Marco

PS: Here's the complete log

```
[18:12:29]Exception (29):
[18:12:29]epc1=0x4022551e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[18:12:29]
[18:12:29]ctx: sys 
[18:12:29]sp: 3fffeae0 end: 3fffffb0 offset: 01a0
[18:12:29]
[18:12:29]>>>stack>>>
[18:12:29]3fffec80:  0000000c 00000019 00000c19 3fff3058  
[18:12:29]3fffec90:  00000010 00000010 00000000 40100575  
[18:12:29]3fffeca0:  00000c17 3fff3058 3fff9774 00000030  
[18:12:29]3fffecb0:  00000010 00000010 00000000 3ffe9e74  
[18:12:29]3fffecc0:  00000010 3fffed20 3fffed20 40229813  
[18:12:29]3fffecd0:  3fff3058 00000abb 00000abb 4010020c  
[18:12:29]3fffece0:  3ffe9e18 3fff96e4 3fffed20 3ffe9e74  
[18:12:29]3fffecf0:  3ffe9e18 3fff96e4 00000098 3ffe9e74  
[18:12:29]3fffed00:  3fffed20 3fff96e4 00000000 402258f3  
[18:12:29]3fffed10:  3ffe9e18 3fff96e4 00000000 4021703f  
[18:12:29]3fffed20:  00000000 00000000 00000000 4010020c  
[18:12:29]3fffed30:  3fff3058 00000c0e 00000c0e 4010020c  
[18:12:29]3fffed40:  3fff9914 3fff96e4 3fff98cc 401006dc  
[18:12:29]3fffed50:  3fff3058 00000c17 00000c17 4010020c  
[18:12:29]3fffed60:  00000000 00000000 3fff9914 401006dc  
[18:12:29]3fffed70:  3fff9914 3fffed90 3fff9914 3ffed1a5  
[18:12:29]3fffed80:  0000002b 3fff96e4 3fff96e4 4021716c  
[18:12:29]3fffed90:  00000000 00000000 00000000 402263ac  
[18:12:29]3fffeda0:  3fff96f4 3fff96e4 3fff59e4 40226710  
[18:12:29]3fffedb0:  40252ae9 00000000 3fff9718 3ffed1a5  
[18:12:29]3fffedc0:  0000002b 0000004d 3fff96e4 40224b04  
[18:12:29]3fffedd0:  0000002b 0000004d 3fff9718 40229c58  
[18:12:29]3fffede0:  3ffed1a5 3fff6bd4 3fff364c 3ffed1a5  
[18:12:29]3fffedf0:  0000002b 0000004d 3fff96e4 40224d9a  
[18:12:29]3fffee00:  00000000 00000000 4bc6a7f0 00000000  
[18:12:29]3fffee10:  00000000 3ffea5d0 4010677c 00001d6b  
[18:12:29]3fffee20:  00000000 00000000 3fff9718 4010020c  
[18:12:29]3fffee30:  002203e1 000010da 000010da 00000000  
[18:12:29]3fffee40:  3fff9484 3fff8a74 00000000 40224dc9  
[18:12:29]3fffee50:  3fff9484 3fff8a74 3fff3650 402228b0  
[18:12:29]3fffee60:  3fff6bd4 3fff2790 3fff6bd4 4024d2f1  
[18:12:29]3fffee70:  3fff36f0 3fff36fc 00000006 3fff364c  
[18:12:29]3fffee80:  3fff6bd4 3fff3654 3fff3650 402228f4  
[18:12:29]3fffee90:  3fff6bd4 3fff3654 3fff3650 40252ab8  
[18:12:29]3fffeea0:  00000094 2901a8c0 00000018 00000094  
[18:12:29]3fffeeb0:  00000000 00000018 3fff0298 40106b0c  
[18:12:29]3fffeec0:  40230000 00000000 00000010 3fff36f4  
[18:12:29]3fffeed0:  3ffed136 3fff36fc 3fff8a74 40250ed5  
[18:12:29]3fffeee0:  3fff3560 3fff55dc 3fff55dc 3fff1698  
[18:12:29]3fffeef0:  00000000 0023c126 402455ef 3fff55dc  
[18:12:29]3fffef00:  3ffed128 00000000 3fff8a74 402502d9  
[... 255 lines cut ...]
[18:12:30]3fffff00:  00000000 3fffdad0 3fff2f18 00000030  
[18:12:30]3fffff10:  00000000 4bc6a7f0 3fff67ec 4022063b  
[18:12:30]3fffff20:  3fff6af0 3fff273c 3fff686c 4021d835  
[18:12:30]3fffff30:  00000000 4bc6a7f0 53333333 00000000  
[18:12:30]3fffff40:  00000000 4bc6a7f0 5a1cac08 00000000  
[18:12:30]3fffff50:  00000000 00000000 4bc6a7f0 00000000  
[18:12:30]3fffff60:  40255b1a 3fff2a84 4010677c 00001d55  
[18:12:30]3fffff70:  00000000 00000000 3ffe895c 00000000  
[18:12:30]3fffff80:  3fffdad0 3fff2f18 4022a3e4 3fffefb0  
[18:12:30]3fffff90:  3fffdad0 00000000 3fff29b8 4022a49e  
[18:12:30]3fffffa0:  3fffdad0 00000000 3fff2a84 4021e0e8  
[18:12:30]<<<stack<<<
[18:12:30]
[18:12:30]last failed alloc call: 3FFEAA9C(152)
[18:12:31]
[18:12:31] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[18:12:31]
[18:12:31]load 0x4010f000, len 1384, room 16 
[18:12:31]tail 8
[18:12:31]chksum 0x2d
[18:12:31]csum 0x2d
[18:12:31]vbb28d4a3
[18:12:31]~ld

```

Aircoookie, 2020-04-16:
@antivirus68 Exception 29 often happens when the flash storing WiFi parameters becomes corrupted. You could try using `esptool.py erase_flash` to reset it and then reflash WLED, it should solve the issue :) 

I still don't fully understand the source of exception 28 that only seems to affect the esp01 builds, but your issue is very likely a different one with invalid flash contents.

antivirus68, 2020-04-17:
> @antivirus68 Exception 29 often happens when the flash storing WiFi parameters becomes corrupted. You could try using `esptool.py erase_flash` to reset it and then reflash WLED, it should solve the [...]

Thank you Aircoookie, this solved my problem.

smith844, 2020-04-17:
I now have two working ESP01S modules with 0.9.1 loaded.  
I used esptool to erase the flash,then to be even more certain I loaded a blank 1M bin using esphome-flasher then loaded WLED_0.9.1_ESP8266_1M_full.bin also with esphome-flasher
I deselected all parts on the config page that i didnt need before i put in my wifi details then rebooted. 
for example 
mDNS = blank
Emulate Alexa Device = unselected
buttons on off = unselected
ntp = unselected
I have put in an MQTT server as I am using homeassistant and I can do remainder through that.  I made the mistake of not removing the mDNS entry on one attempt and as soon as it rebooted it boot looped with the following log repeated around 8 seconds:
```
[19:15:19]
[19:15:19]Exception (28):
[19:15:19]epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[19:15:19]
[19:15:19]>>>stack>>>
[19:15:19]
[19:15:19]ctx: cont
[19:15:19]sp: 3ffffd30 end: 3fffffc0 offset: 01a0
[19:15:19]3ffffed0:  00000000 3fffc6fc 00000001 3fff2398  
[19:15:19]3ffffee0:  3fffff40 3fffff4c 00000000 402228e3  
[19:15:19]3ffffef0:  00000050 0000ff45 40100b30 3fff2170  
[19:15:19]3fffff00:  00000003 3fff2378 3fff184c 3fffff4c  
[19:15:19]3fffff10:  3fffff40 00000050 3fff2398 4022294e  
[19:15:19]3fffff20:  3ffeabd3 3fffff40 3ffeabcf 3fff2170  
[19:15:19]3fffff30:  3ffeabd3 3ffeabcf 3fff2398 4021c07e  
[19:15:19]3fffff40:  00706374 1601a8c0 83c6a7f0 64656c77  
[19:15:19]3fffff50:  3fff5400 842210bc 40100b30 3fff2170  
[19:15:19]3fffff60:  3fff2175 3fff216f 00000000 4021c2de  
[19:15:19]3fffff70:  00000000 402210bc 3fff2598 3fff25d8  
[19:15:19]3fffff80:  00000000 00000000 00000001 3fff25d8  
[19:15:19]3fffff90:  3fffdad0 00000000 3fff2598 4021f3e7  
[19:15:19]3fffffa0:  feefeffe feefeffe feefeffe 402304c4  
[19:15:19]3fffffb0:  feefeffe feefeffe 3ffe8800 4010089d  
[19:15:19]<<<stack<<<
[19:15:19]
[19:15:19] ets Jan  8 2013,rst cause:2, boot mode:(3,7)
[19:15:19]
[19:15:19]load 0x4010f000, len 1384, room 16 
[19:15:19]tail 8
[19:15:19]chksum 0x2d
[19:15:19]csum 0x2d
[19:15:19]vbc204a9b
[19:15:19]~ld
[19:15:19]Ada
```
But I am happy it is working, i would recommend deselecting all the options you are not going to need.  It is safer that way.

ClintH, 2020-04-24:
> @ClintH your symptoms are not the same as this thread, as this thread is about bootloop after setting wifi details and reboot. Are you sure you have a good powersupply (not all breadboards are 100%, [...]

OK, yes it is a different issue. I'll wait for posting a new issue until I have a new UART USB adapter. Right now the one I have is hardwired for flashing.

iMartyn, 2020-05-13:
I can also confirm with a Wemos D1 Mini (standard 8266 version) mDNS filled in bootloops on 0.10.0_ESP8266_1M_full still, empty string and it's fine.

aenertia, 2020-06-22:
Adding here git compiled 0.10.0 (commit f8266d98117e933b63a71e5a7a94603ec8498cdb) using the nodemcu2 profile with LED pin 3 on a Sonoff Basic with 4MB spi mod illicits the same boot loop behaviour after entering wifi credentials.

aenertia, 2020-06-22:
just an FYI blanking mDNS field disables it. Given that seems to fix it for some % of users it's likely a race condition ; also given that on some wifi networks it isn't triggered (android AP i.e as per one post) it's likely it is an interaction with mDNS implementations on the routed network with the esp mDNS library. ... I am running avahi with openwrt and it does trigger the issue.

davidboon78, 2020-08-02:
I had this problem with my ESP01 1M.  Switched to 0.8.6 and it works fine.  Would like to be able to run the newer features though so I'm following this thread in the hopes a future version will be tested to work with the ESP01.   Specifically I want to be able to run them as ArtNet DMX lights as well.  Might make a fork with that working feature cause WLED is a pretty amazing system, hats off to AirCookie for such a comprehensive smart light.

garci66, 2020-09-16:
I'll add a +1 on an ESP8266 (sonoff mini R1) with 1MByte of flash getting into a bootloop with mdns enabled. Actually, none of the custom builds worked out of the box for me (I had to build my own - default settings, but DEBUG enabled) on gitpod). Otherwise, the WLED_AP wouldnt even show up.

im running my own build of 0.10.2 at the moment. Just getting started.

Aircoookie, 2020-09-17:
@garci66 the issue has been fixed :) Try building the latest master (or download the 1m_full binary of 0.10.2 which I've updated to use the fix)

garci66, 2020-09-17:
HI @Aircoookie indeed the first binary I tried was: WLED_0.10.2_ESP01_1M_full_fix2009030.bin and it never published the SSID. WLED was working cause if I pressed the button on Gpio0 it would toggle colors / turn on / off, but I never saw the SSID. And when I built from master using debug, I saw that the console was showing "Wifi status: 0" on every regular printout. When building I was getting a warning about two different versions  of the espressif sdk or something similar. (I was building on gitpod with platformio)

garci66, 2020-09-17:
Im quite flexible with testing (so for example, one of my esp, im fairly certain the serial RX died when connecting to the led strips so recovered it using a flash programmer and writing directly to the SPI flash). The serial is still dead but the device works as a sonoff basic so its not all lost.

Even on my own builds, there seems to be some sort of race condition, as things only seem to work on debug builds (or rather, builds with debug DEFINES (but not build_type=debug). Exact same build but without debug strings I never get the WLED-AP to show up.

FulvioSpelta, 2020-12-26:
Hi all, I kindly ask the community for an update or hint on this topic.
I've found the same problem, that means for me: after flash the AP WLED sometimes is present but it go off in few seconds.
If i push the button the relè switchs on/off
The hw is a sonoff basic R1

I've checked these "stock" versions:
"WLED_0.10.2_ESP01_1M_full_fix2009030"
"WLED_0.11.1_ESP8266_1M"

I'm available for support in bug finding but i need hint about what version should be compiled and how to activate debug.
Sorry but i'm a novice with this environment and sw

Thanks

Amli2011, 2020-12-26:
I have tested WLED_0.11.1_ESP8266_1M now and the bug in this thread is fixed. 
Thank you!

FulvioSpelta: I think you are describing another problem. I suggest you capture the output on the serial tx/rx pins of the device when it boots. Check for hints of what is wrong, and if you don't find the solution then open a new issue with a copy of the serial output, and info about the problem and your hardware setup. DO NOT CONNECT TO LIVE (230V/110V) WHEN SERIAL IS CONNECTED - only use 3.3V from the serial interface.

closed 2020-12-26 by Amli2011