wled/WLED #462: Integration with DMX

raresserban, 2019-12-15:
Hello,
I want to make a usermod to integrate this library with a DMX light, similar to this:
https://www.youtube.com/watch?v=4PjBBBQB2m4

What I was thinking is to have the light as another pixel in the strip, but I am not sure how I can do that. If that is not possible, then I would like at least for the effects to be applied to the DMX light as well.

How can I have access in the `userLoop` function to the current effect, or the value of a pixel from the strip so I can update my DMX light?

Aircoookie, 2019-12-15:
Hi!
That sounds like a really cool idea! You can get the color of any pixel quite easily:
Just use `strip.getPixelColor(i)`. That will give you the color in uint32_t format. If you need the R,G,B components, you can do:
```cpp
uint16_t pixel = 0;
uint32_t in = strip.getPixelColor(pixel);
byte r = in >> 16 & 0xFF;
byte g = in >> 8  & 0xFF;
byte b = in       & 0xFF;
```

Looking forward to what you come up with and let me know if you need anything else :)

jwingefeld, 2020-01-29:
> I want to make a usermod to integrate this library with a DMX light, similar to this:
> https://www.youtube.com/watch?v=4PjBBBQB2m4

Hello raresserban,

i am currently working on this. It already works, i just have to make it "user servicable". Let me know if you want to try it out.

/Jan

raresserban, 2020-01-30:
@jwingefeld It would be great if you could share the code!I tried something but with not luck so currently I am waiting for a new DMX cable, didn't have time to do this for christmas.

jwingefeld, 2020-01-30:
I will let you know when it is usable. Should be a matter of days.

Please note that you will need a MAX485 chip connected to your ESP8266.

abyssdj, 2020-02-04:
> 
> 
> I will let you know when it is usable. Should be a matter of days.
> 
> Please note that you will need a MAX485 chip connected to your ESP8266.

I would also be interested in this. I've recently discovered WLED and it fits my requirements perfectly for LED strips, but would also be great to have a DMX counterpart, so that all the nodes use the same software. Extra bonuses would be "effects" for pan/tilt, etc to be able to run a basic display as a fallback when not receiving sacn.

abyssdj, 2020-02-06:
Extending on the above, would it be possible to have Artnet as an alternative option to sACN? I find it faster and more suitable in my case.

Aircoookie, 2020-02-06:
@abyssdj yes, ArtNet support has also already been requested in https://github.com/Aircoookie/WLED/issues/417

abyssdj, 2020-02-07:
> 
> 
> @abyssdj yes, ArtNet support has also already been requested in #417

Perfect!

jwingefeld, 2020-03-04:
It's done! It works and it's in WLED.
To try DMX output support, you'll need to grab the latest source (not binary) from github and compile it yourself. Further instructions are here: https://github.com/Aircoookie/WLED/wiki/DMX-Output. Let me know if you need any help or if the instructions are unclear.

raresserban, 2020-03-24:
@jwingefeld Could you help me with debugging some DMX issues?
I have used this schema to connect the MAX485 to NodeMCU, and used D4 pin (since in code it is pin #2, so D4 on NodeMDCU), but also tried TX pin, and used the example:
https://github.com/Rickgg/ESP-Dmx/blob/master/examples/DMX_Send/DMX_Send.ino, looked at my light to configure it properly, but nothing works and the light just stays off.

I got a DMX cable off of Aliexpress that I cut in half and put some Dupont cables on one end. Could this cause signal degradations so my light doesn't get the signal?The whole cable is like 75cm in length.

raresserban, 2020-03-24:
Thanks @jwingefeld, finally had the change to test it after I got my DMX cable and it works beautifully!

Now to have WLED support both DMX and normal leds, and have the DMX light as a segment :D
I think I am going to work on this and maybe submit a MR if it works properly.

closed 2020-03-24 by raresserban

abyssdj, 2020-04-11:
I had a chance to mess around with this, but found it reasonably unresponsive for some reason. I found that it would often miss colour changes, etc. I'll experiment some more. That may be something at my end, but either way, what I would like to see is a way to specify a range of channels that are just dmx output, just (for example) plain artnet-dmx.  Would this be possible?

As an example of the setup, segment 1 may be channels 1-180 may be set as 60x rgb led's, followed by segment 2 using 181-200 as basic dmx channels, for controlling moving heads, etc. 

"Advanced" functions that would be nice to see would be the ability to set a channel to "random" (for pan/tilt - I did modify a few lines very quickly to try this, it kind of worked) or set channels which would cover rgb fixtures to apply colour 1/2/3 of the pallete of the current preset, and so on. That way, it could be a very nice, basic standalone setup. The fact that it would fall back to this within seconds when SACN goes down would make it great as a backup.

harueg, 2020-11-26:
Hi @jwingefeld 

I tried to compile it for an AZ Delivery ESP-32 Dev Kit C V4  but didn't work, is rebooting constantly.
I've set DMX sendpin to gpio23, ledpin is at gpio2
the downloaded binary itself works.
The source of the wled 0.10.2 has the dmx already enabled, but I can't see any dmx button on the UI.
@Aircoookie Maybe it is not enabled in the binary ?

What other settings might be necessary ?

Thank you

Harald