wled/WLED #463: Can't sync LEDs strip with other IP in same network together

tihoangyeudau, 2019-12-15:
When I press the sync button of the led strip has ip: 192.168.0.100, then select an effect, the remaining led strip has ip: 192.168.0.102 just blinks and turns off without changing the effect according to main led.

i try to change numled of all led strip to same 300 but not work

(d1 mini, 921600, 80mhz, 1M, v2 HB, Easer flash all content, 300 leds)

Aircoookie, 2019-12-15:
Hi and thanks for reporting!
That's a very strange issue! Please first check if all these checkboxes are enabled in Sync settings :)
[image]

You can also try compilinh with lwip 1.4 HB, that's what I use, maybe it helps!

tihoangyeudau, 2019-12-16:
It works. Thank you!

closed 2019-12-16 by tihoangyeudau