wled/WLED #464: Save/restore current effect after power off

pruwait, 2019-12-15:
Very good project!

Ideas to do:
Save/restore current color palette & effect in EEPROM
Now it works with web power button only. Need restore memory on start ESP8266.

Add ModbusRTU cotrol (WLED is modbus slave)

Aircoookie, 2019-12-15:
Hi!
Awesome, great that you like it!
Saving the current effect config is supported via presets (the 1-16 numbers in favorites tab)!
You can also set one of them to be applied on start with the "Boot preset" option in LED settings :)

Ok, I will think about Modbus. What controller are you interested in using together with WLED?

pruwait, 2019-12-25:
I use Mini NodeMcu like WLED controller + hassOS on odroid XU4 like modbus master+ XY-485 board.