wled/WLED #465: Transitions don't work with multiple segemnts (0.90.b1)

badbadc0ffee, 2019-12-15:
It looks like the transitions between effects don't work any longer when I use segements. When the while strip is used as one segment, the same transitions as seen with 0.8x versions work fine.

Aircoookie, 2019-12-15:
Hi and thanks for reporting this issue!
Yes that is correct, transitions currently only work on brightness and the "main segment", which is Segment 0 by default. Supporting transitions for all segments would cost too much memory with the current implementation, but I already got something in mind to make it work soon :)

bgberk, 2020-02-04:
Is there any update or insight into how this implementation might work? I'm looking to wire up a series of niches that can fade between colors or off and on independently and it seems like segments is the best way to do that (without needing a separate device controlling each niche).

xBelladonna, 2020-11-23:
First of all, thank you very much for providing such a great project and absolutely stellar support to go with it! I'm also interested in overview/details of the implementation, since this limitation is currently the only thing preventing me from using segments as I value brightness transitions over segment control.

blazoncek, 2021-10-03:
This is supposedly fixed in 0.13-b3. Closing.
Feel free to reopen if the issue persists.

closed 2021-10-03 by blazoncek