wled/WLED #466: (Some?) effects appear to be running in selected segments and whole strip simultaneously (0.9.0-b1)

badbadc0ffee, 2019-12-15:


raoulteeuwen, 2019-12-15:
While i was testing segments, trying to put different effects in different segments, changing effects, and trying to switch back to the strip being 1 segment, i also had various 'struggles'/interesting effects, amongst which was sometimes 2 effects seemed to run on 1 segment

Aircoookie, 2019-12-15:
Thank you for the report! I indeed just tried all effects and a few are not behaving correctly!

- Colorful (turns on up to 3 additional LEDs after the segment end)
- Loading and Gradient (those turn on LEDs after the segment to static?)
- Sweep/Sweep random (the "back sweep" happens on the first LEDs of the strip rather than in segment)

Will get them fixed soon! Please let me know if you noticed weird behavior on other effects. Also note that overlaying segments (when the ranges overlap) is not supported currently and might lead to weirdness :)

closed 2020-01-04 by Aircoookie