wled/WLED #468: board is resetting all the time

pontos-dev, 2019-12-15:
After update to 0.9b my board has began to reset/reconneting all the time ( I can see the wled board on my router's client list, but it is there 5 sec. ish, and then disappears around 5 sec. and so it appears again, again and again.... 

So I can't do nothing with it
I seeing forward to try the new update, I can see folks love it, and I really love the 0.8.6 version, so I can only be good :)
Keep up the good work Aircoookie :)

n1iic, 2019-12-15:
I'll second that. I am running an Amica board amazon.com/gp/product/B07S5Z3VYZ and have the same problem with 0.8.6 and above. 0.8.5 works like a charm.
I also have the same sentiment of holy crap, what a lot of work over time and everyone should be saying thank you! I read about 0.9 and it sounds fantastic, and I look forward to trying it soon.

Mariu86, 2019-12-16:
0.9 available and it's working ok , it's a pre-relese.

DonEugraf, 2019-12-16:
I'm using it with Hyperion and I love the new look on 0.9but it keeps rebooting every minute or so.
Keep up this amazing project!

jeffchampagne, 2019-12-17:
Same problem here with 0.9b on an ESP8266. The board resets if I adjust too many settings in the UI. I haven't been able to determine a pattern yet beyond just changing an effect or color a couple of times. I also get periodic JSON submission errors before the board resets, but not every time.

mike811, 2019-12-17:
Same here with wemos d1 - constant reboot after initial wlan setup (my ssid etc.) with _1M_full.bin
No reboots with _ESP8266.bin though.
Unfortunately not very stable this release. Sometimes freezes and no control possible.

Aircoookie, 2020-01-04:
Sorry for all the connectivity issues! While I haven't been able to reproduce these myself, I will try to release a new version soon that may hopefully alleviate some of them!

pontos-dev, 2020-01-04:
It sounds good aircoookie, I absolutely Love your work you have done with
the wled firmware/software

I will wait with longing :)

ggrant111, 2020-01-28:
Any word on this issue as of late? I have had a few D1mini spares do the constant reboot after I try to connect it to my home network. If I use the AP mode it doesn’t do this. I am powering it from the USB port not sure if this is causing the issue but haven’t experienced it until the last batch I loaded the wled firmware on. Running 9.0 on them.

sansillusion, 2020-01-29:
This sound like common bad usb cable/power source

ggrant111, 2020-01-29:
I have changed the cable power supply etc. it works fine before adding it
the my local network. So if I use it in AP mode it’s fine. Only way I can
get it to get out of the cycle mode is to reflash the firmware via usb and
radio Ide or espeasy

pontos-dev, 2020-05-29:
I still have the same issue as of 1.9.0 firmware, and have tryed with 3 different d1 mini's

Legsmaniac, 2020-05-29:
Looks like I was not the only one who had (recently) problems with D1 Minis.
@pontos-dev can I ask which make/models they were?  Mine were [these](https://www.amazon.co.uk/gp/product/B083M6N55Z/) and I had problems with them.  I tried 4 of them and I returned all for a refund.

pontos-dev, 2020-05-30:
No, sadly enough, it's very annoying, [image] it's these i have, not the pro versions

Legsmaniac, 2020-05-30:
Wow, they're cheap. Maybe they're cheap crap? That's what I reckon mine were. Duff cheap copies.  I guess there's something to be said about paying a bit more for quality?

pontos-dev, 2020-05-30:
Very cheap, I bourght them because og the price, but one of them works, but with 0.9 on it, i will not true to update it, it's my moms light I gave her in birthday gift, so it shall always work :) 
But i have considered buying a esp32 instead

Aircoookie, 2020-05-31:
@pontos-dev can you try this binary? Maybe it helps :) https://github.com/Aircoookie/WLED/issues/946#issuecomment-634100863

If it still bootloops, you can try disabling the Alexa emulation in Sync settings before putting in your home network info.

closed 2020-10-07 by stale[bot]