wled/WLED #469: Add current IP address MQTT topic to JSON state

joecotton, 2019-12-16:
I would like to see the IP Address and MQTT topic(s) in the JSON state that is returned by http://x.x.x.x/json

jsenecal, 2019-12-20:
Why the IP address though? Don't you already use/know it in/from the url ?

joecotton, 2019-12-20:
The current way to get/send information to WLED is a mix of MQTT messages and HTTP requests. For MQTT, you don't need to know the IP, but you do need to know the topic.

I am using Node-Red to control WLED, and Node-Red is tightly integrated with MQTT and JSON. If one subscribes in Node-Red to the WLED topic, then to send JSON settings updates to WLED is through an HTTP request to http://x.x.x.x/json.

The method of setting configs through partial JSON state looks like an excellent way to control WLED, but there doesn't appear to be a way to send JSON to WLED via MQTT. So it would be helpful to be able to extract the IP address if initial communication is via MQTT. 

Alternately, WLED could accept JSON state updates though MQTT, rather than only accepting the HTTP-style API through the `wled/[topic]/api` topic. In that case, discovering the IP is not as important.

But it may be useful for debugging in an MQTT environment to know the IP addresses of multiple WLED controllers on the network.

To also send the MQTT topic(s) via the HTTP JSON state would just fill in a gap in completeness on device config info available by either method.

Aircoookie, 2022-10-21:
IP is added to `/json/info` :) An MQTT topic is not planned, but JSON info might be available via MQTT at some point.

closed 2022-10-21 by Aircoookie