wled/WLED #470: [Feature request] Include Color-Fading in Nightlight

Def3nder, 2019-12-16:
Hi Aircoookie,

I use the nightlight feature to simulate a sunrise with the ceiling LED strips in the bed room with a macro (CL=h000000&W=255&A=5&NL=10&NT=128&NF=1) to raise the brightness of the white channel from 2% to 50%. 

A further improvement would be having the color tone changed, too.
So we could simulate a real sunrise.

If we could add a target color, we could set the actual color to reddish and the target color to warm white (or yellow) and then nightlight would fade both brightness and color.

What do you think ?

eliobou, 2019-12-17:
A sunrise mode would be awesome as I use my WLED to wake up

Aircoookie, 2020-01-04:
Hi!
Target color would definitely be possible, will think about that!
Also, hopefully find some time to add https://github.com/Aircoookie/WLED/issues/311 soon :)

Def3nder, 2020-01-22:
implemented with #571 - Thx !

closed 2020-01-22 by Def3nder

eliobou, 2020-02-23:
Can you please explain how to use it ? I tried with a macro but have a strange result, I'm probably doing it wrong. What would be the macro if I want let say red to white transition over 10mn ?