wled/WLED #471: 0.9.0 - Some Effects Gone - Like Out-In?

sr1sws, 2019-12-16:
It seems like some of the effects are missing in 0.9.0-b1? Like Out-In, In-Out, etc?... or is there another way to achieve the effects that I'm just not clever enough to realize?

OTOH... I really like the interface, the alphabetized list of effects and probably "segments" as soon as I figure them out. ;)

Thanks for your continued efforts on this software!

Steve

Aircoookie, 2019-12-16:
Hi Steve,

yep, I've removed those as they took up a lot of flash.
You can easily replicate them using the "Wipe" mode and two segments :)

Let me know if you have any questions!

Christian

sr1sws, 2019-12-16:
Thanks! I kinda figured there was an alternate way. I'll have to play with the segments.

If I set up an effect like this with two segments, can that be stored in a preset?

Many thanks and again, I think this is pretty cool software!
Steve

brentbrooks70, 2019-12-16:
@sr1sws preset 16 seems capable of saving segments. I have 1 device saving 8 segments to this preset

sr1sws, 2019-12-16:
Thank you. I guess that falls a bit short... I'm using a couple of those effects for Christmas, saved in presets and then cycled through. But there are probably other effects that would work too.  Thanks for the tip!
Steve

Aircoookie, 2019-12-16:
@sr1sws I agree that just Preset 16 isn't ideal. Some of the next updates will include a higher number of supported presets and all of them will support segments! What you can do for now is save preset 16 with both segments selected (The little checkmark left of the segment number), in that case a preset cycle will apply the effects to both segments. Might not help though since you might have some effects where you want the whole strip to be one segment.

sr1sws, 2019-12-17:
Thanks... I did just play with setting up 2 segments and saved the config to preset one.  Went and ran a different set of effects and then loaded from preset one - both segments appeared to restore OK. ;) 

Thank you!

sr1sws, 2019-12-17:
Yeah... OK... it's not working exactly like I thought. Would love to be able to have the complete config save to any preset - but I'm sure you realize that.

I still love the software. And the 'glitter' effect is very cool.

Steve

Aircoookie, 2020-01-04:
Of course, I am sorry for the inconvenience! A better preset system will be introduced soon :)

closed 2020-01-04 by Aircoookie