wled/WLED #474: 0.9.0-b1 led count is ignored

huggy-d1, 2019-12-16:
In testing a setup with low 12V power, I only want to test 45 LEDs at full power.  My strip is much longer, like 250 LEDs (have not unrolled it all to count yet).
Configured WLED LED count to 45 but it is controlling the entire strip no matter what number I insert in that field.
Luckily I had a reasonable low current limit or I would have burned up the power supply trying to run the entire strip at full power.

Aircoookie, 2019-12-16:
Hi! If you set 45 as LED count and save, then go back and enter the LED settings again, does it still say 45? You can also try to change the color and see if it applies to all LEDs. If you change the LED count to something lower, you might need to power-cycle for the LEDs past 45 to actually turn off.

brentbrooks70, 2019-12-16:
@huggy-d1 What kind of strip is it? Most 12V strips control 3 LEDs per "pixel" so setting 45 LEDs could possibly be actually controlling 135.

huggy-d1, 2019-12-16:
@Aircoookie changing the LED count is not live, nor is it something that affects the strip after a power cycle either.  It comes up and shows 45.  I even went down to 15, saved, changed effect, power cycled, and no difference.

Aircoookie, 2020-01-04:
@huggy-d1 sorry for taking long to get back to you on this. What is the exact type of LED strip you are using?

huggy-d1, 2020-01-06:
12V addressable RGB, https://www.amazon.com/dp/B07JPXHK5L
It seems to work other than the max LED part.
Once the system is installed and running, I won't care anymore because there will not be extra LEDs at the end of the strip not being used.
However, during testing to simulate being installed, it behaved in a way that did not make sense.

raoulteeuwen, 2020-01-06:
@huggy-d1 fyi, the LED count works perfectly for me, so it should be a matter of config/hw etc... What controller do you use exactly, with what bin exactly? Do other values you change in WLED persist after reboot?

huggy-d1, 2020-01-08:
@raoulteeuwen are you saying that if your strip has 300 LEDs (as an example), and you only want to test say 30, you can insert the value 30 for the number of LEDs and only 30 come on and are managed by WLED?

My issue is that regardless of the value I insert for the count, somehow WLED figures out how many LEDs are in the strip and controls every single one of them.

I know because I selected one of the patterns that does that. It illuminated every LED in the strip and came back after hitting the end.

How could it know or default to the _exact_ correct value?  *I* do not even know how many LEDs are on the strip.  I never counted.

The wifi settings persist.  The device name persists.  The power up color persists.

raoulteeuwen, 2020-01-08:
Yes @huggy-d1 , that's (put a nr in that field and only that nr of LEDs light up, like it is a virtual strip length) how it is working for me, and afaik is the way it is designed to work. Afaik, there is no protocol to determine the nr of LEDs on a strip.

huggy-d1, 2020-01-08:
@raoulteeuwen I *know* there isn't a protocol to determine the actual strip length, and I know @Aircoookie isn't that good to precisely guess the exact number of LEDs on my strip.

I just found it quite odd that no matter how I saved the led count, from 1, 15, 30, 45, 100, the behavior of the strip was identical.  I tried save, save / reboot, save, power cycle, save / reboot / power cycle.  

All had exactly the same strange results.
And... I only had one WLED powered up.  The UI would power the strip on / off as expected and I could set the color / brightness / speed, so I know my UI was the correct one (in case I somehow magically reconfigured a neighbor's WLED setup at the exact moment they were configuring theirs!)...
Yes, I thought of all the odd things that could result in what I saw, and I am at a loss for why it misbehaved.

closed 2020-05-14 by stale[bot]

codefaux, 2021-04-10:
For future diggers - I was having a similar issue and I found out where the mistake is. You must change your LED count in two places.  "Total LED Count" AT THE TOP -- which I somehow missed -- _and_ in the hardware configuration of the LEDs/pins/etc.