wled/WLED #475: Color/Brightness settings lost after power loss, restore favorites not working

freemann, 2019-12-17:
I'm running the current Beta "WLED version 0.9.0-b1".
After setting stuff like Color palet, Effect Speed, Effect intensity, Brightness, etc. and I disconnect the ESP8266MOD from power, the settings are lost.

I tried to save the settings to a Favorites Slot and restore them, but thats for some reason not working.

I'm i doing something wrong or .... ?

Aircoookie, 2019-12-19:
Hi and sorry for the issues!
What tool and command did you use to flash the ESP? It sounds like the flash layout is incorrect so that settings can't be used...

Happy holidays!

freemann, 2019-12-20:
Hi, first I tried flashing through the Arduino IDE.
Then I screwed things up, the device was inaccessible, had to reflash it and used the http minimal flasher to flash 0.8.6. After that I used the WLED interface to flash "WLED_0.9.0-b1_ESP8266_1M_full.bin" on the device.

Is this the right way to go or should I use a different flash/update path?

Aircoookie, 2019-12-21:
That is the right way, but the binary you used is intended for ESP01 modules with 1mb of flash. Try the regular `WLED_0.9.0-b1_ESP8266.bin` (flash minimal flasher and directly flash it from there), that may work better :)

closed 2020-05-10 by stale[bot]