wled/WLED #479: Backing up your Nodemcu to a file

rfordhamjr, 2019-12-18:
I have been using ESPHome-Flasher to flash my Nodemcu. Now that I have everything set the way I want it. Is there an app that will save it to a file to keep as a backup?

sr1sws, 2019-12-18:
Read my mind. I came here to ask for a way, or if there is a way to export at least the presets. Would be nice to have sets of preset files for different holidays. A backup of the entire config would help too.

Aircoookie, 2019-12-19:
In one of the next versions, you will be able to simply export `presets.json` and `config.json` files.
For now, you can read back the flash and save it to a bin file you could restore with either ESPHomeflasher or esptool. You will need to install [esptool](https://github.com/espressif/esptool), then run (replace PORT with the com port of your ESP, remove `-p PORT` if you don't know):
```
esptool.py -p PORT -b 460800 read_flash 0 0x400000 wled_backup.bin
```

Hope that helps :)

sr1sws, 2019-12-19:
Great tip, thank you!

Morphy99, 2020-08-14:
FYI [Tasmotizer](https://github.com/tasmota/tasmotizer/issues/39#issuecomment-674069258) *should* do this but it doesn't seem to be working for me on Windows.

blazoncek, 2022-02-21:
Backup & Restore available since 0.13-b4 under Security & Updates settings.
Closing.

closed 2022-02-21 by blazoncek