wled/WLED #482: 0.9.0-b1 - "Set current preset cycle setting as boot default" Doesn't seem to stick on save?

chunkysteveo, 2019-12-18:
Unsure if this is common for all, but running 0.9.0-b1 ESP8266-1Mb-OTA Bin file on an ESP-01 and when I turn off and on one of my controllers (that syncs to the rest in the house), the preset cycle I had saves does not run, it seems to run the last effect it was on.

Checking LED preferences menu and option "Set current preset cycle setting as boot default:" does not seem to persist if I tick it, leave the menu and back in. I suspect that is why it's not running my preset loop on "power on"?

Thanks for ALL your hard work so far @Aircoookie !

Steve

sansillusion, 2019-12-18:
It's only at boot (power unplugged)
I dont know if there is a way to get a preset after software power on.

UbiMaiden, 2019-12-18:
I have the same issue, it just stay on one preset, no cycle. I tried Power on/off with the same result. Sometimes I get JSON error when, I try to setup cycle under Favorites.
All I need is to cycle the preset when I power on the WLED controller, no fancy setup.

jarrah31, 2019-12-18:
I was going to raise this as an issue too because the problem exists on 0.8.6 as well as 0.9.0-b1.
Steps to reproduce:

1. Save effects to favourite slots
2. Set first and last preset (e.g. 1 to 14)
3. Enable "preset cycle" tick box. Lights cycle through presets as expected.
4. Goto Config -> LED Preferences
5. Enable the "Set current preset cycle setting as boot default" tick box
6. Press Save
7. Press the Power button on the main home page to turn off the lights, then turn them back on.
8. Lights return to last preset and no longer cycles through presets.

I use WLED for external Christmas lights that I turn on/off at sunset/sunrise via Home Assistant, but each day I arrive home I have to manually re-enable preset cycling.

My thanks too for all your hard work Aircookie!!

chunkysteveo, 2019-12-18:
Ahhh @sansillusion , I see. I am software powering it off and on with Home Assistant  like @jarrah31 . There needs to be a way that the software off/on is like booting and the controller resumes what it was doing? The tick box never stays ticked for me. Not tried a physical power reboot yet to see if the preset cycle resumes.

sr1sws, 2019-12-18:
I haven't tried this with 0.9.0, but in 0.8.6, I use the timer and macro functions. I have my come on at 5:30pm and off at 11:59pm. 5:30pm is close enough to dusk for my purposes.  I also have the macro set as the default at power up... so if power drops during the day, the LEDs are on - not a big deal to me.

Hope this helps.
Steve

UbiMaiden, 2019-12-19:
I tried Power ON/OFF the WLED controller, but I still need to go into favorites and tick the power cycle box, If I press anything else cycle doesn't work.

wchpikus, 2019-12-20:
The same, after reboot cirxle do not work.

timdonovanuk, 2019-12-20:
Same for me. Ticked the box, rebooted, and the setting is lost.

chunkysteveo, 2019-12-20:
Looking like this is a 'feature', not a bug as such. May be worth looking into the next release - on soft (or hard) boot the controller resumes the preset cycle if the tickbox is checked in the settings @Aircoookie - if possible? For Christmas decorations that are timed via Home Assistant to come on/go off - this is a must (2020! Lol)

timdonovanuk, 2019-12-21:
Is there a way to trigger this via a REST command? Can be a temporary solution to this issue, can script Home Assitant to send the command when WLED changes state to ON.

timdonovanuk, 2019-12-21:
Found out you can call http://wled.home/win&CY=1 to trigger your playlist! Cheers.

chunkysteveo, 2019-12-22:
Ahh that's a good solution for me @timdonovanuk with Home Assistant. Thanks for the tip!

woody2010, 2019-12-26:
Just to make sure I'm on the same page as the rest of you. There is currently no way to save "SYNC" and preset cycle settings if I'm not using HA right?  I have 3 nodemcu's that sync to one, but I currently have to manually click SYNC and set the preset cycle settings. Thanks

sr1sws, 2019-12-26:
I only have two, but I click "sync" on the one I consider the master and the other does the same sequence of presets, colors, etc. Presets on the 'slave' unit are NOT the same as the master.

I have mine set to power up by the internal clock. I have the slave power up 1 minute early so it can receive the sync commands from the master. 

Notes:
- units are set by default to receive sync commands
- units are set to be in the same group/universe/whatever by default - you can change
- you can sync multiple units - one master, many slaves
- it isn't quite perfect... my slave occasionally does it's own thing... but generally seems to re-sync at some point
- I turned on 'send sync twice' (or whatever the setting is actually called) due to the occasional non-sync
- I don't know how often the sync commands are sent. I kind of assume at the beginning of each change.
- I have 16 presets on the master and cycle thru then via a macro setting. I think the sync command is sent upon each preset change, but I'm not positive.

Hope this helps.

gregw013, 2020-01-02:
I ran into the same issues, and solved it following the method below. I’m currently running 7 ESP8266s for Christmas lights on my house, and they’re powered on via a smart plug and the sync/preset cycles worked perfectly all through this Christmas season. 

On your ‘master’ controller (that your presets are saved on):

Navigate to Settings > Sync Interfaces > WLED Broadcast. Check the box for ‘Send notifications on direct change’. This will default the ‘sync’ switch to the on position. 

Under User > Interface, set the UI mode back to classic. Refresh the page. 

Click the Heart on the top of the screen. On this screen, you can set which presets to cycle through and how long each preset runs for. Select the check box, then hit ‘apply boot config’.

I spent a few days trying to figure this out, and since stumbling across these settings it has been working flawlessly. For some reason the cycle settings have to be enabled via the Classic UI for it to work properly

closed 2020-05-10 by stale[bot]