wled/WLED #483: So hard to flash the binaries for me

wunderbaum, 2019-12-18:
What am I doing wrong or do I have the wrong hardware?
I tried with some wemos-boards and "nacked" ESP8266-boards. I only had success with the precompiled lwip-version.
Flashing other binaries like tasmota went easy and made no trouble.

So, I flash, that works as usual, but it cannot boot.

` ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbc204a9b
~ld
Dr `

Aircoookie, 2019-12-19:
Hi and sorry for the issues!
What tool/command did you use and what binaries/WLED versions did you try?

Happy holidays by the way!

Mariu86, 2019-12-19:
I finally getting compiled latest master. if you use all the settings in the wiki and all the library ( I missed one and I have a lots of errors till now) now I compiled and use the Arduino ota to update  ! For a noob like me this is a great success! Thanks Aircoookie and happy holidays!

wunderbaum, 2019-12-20:
I tried every binary from the last and flashed it with the espeasy flash tool.

wunderbaum, 2019-12-20:
I was able to flash and use the older lwip-version, but none of the newer ones I got to work. I can flash them, but they do not start and open the wlan-configuration AP.

Mariu86, 2019-12-20:
If you want to do it easy way, you can flash the http ota update and upload the binaries, I try this way and it's the easiest way to flash a new device, i try the esptool but I didn't get anywhere maybe I done something wrong.

wunderbaum, 2019-12-21:
yeah!
Thanks alot! Finally it works and I can have my christmas lights.

closed 2019-12-22 by Aircoookie

Mariu86, 2019-12-23:
How do you do the upload ?