wled/WLED #484: What kind of chips are compatible?

steve899999, 2019-12-19:
Looking into starting a project of using LED lighting strips. Was wondering just what kind of chips are compatible to be used with this software? Been looking at a variety ranging from Tm1814, sk6812, Ws2812b, and so on. While I saw that WS2812B is compatible, was wondering just what other chipsets are usable as well with this software. A full listing or simply comparable chipsets would be much MUCh appreciated :)

Def3nder, 2019-12-19:
Hi  @steve899999 , the compatible chipsets for the color-coding are

1 pin:
- WS2812B (5V)
- WS2811 (12V power, with 5V signal)
- WS2813 (WS2812 with redundant data on 2 wires DI and BI, to resist LEDs failure)
- WS2815 (like WS2813 but 12V), send 5V signal on BI.
   - BTF2815 (cheaper 12V)
- SK6813 (redundant like WS2813)
- SK6812 (can support up to 4 colors, commonly GRBW)
   - SK6805 (3 colors)

2 pins (clock CI and data DI) chips:
- APA102 
- SK9822
- WS2801

Mariu86, 2019-12-19:
This it's a good info to put in the wiki !

raoulteeuwen, 2019-12-20:
Added to the wiki

closed 2019-12-21 by Aircoookie

tpmodding, 2020-01-03:
Could someone explain (could not find it) how to set the different LED chips...with the webui or do i have to compile it? and with which pins? thanks

Aircoookie, 2020-01-06:
@tpmodding most use the same WS281x protocol and as such don't need a recompile! For APA102, there is a dedicated binary. For other chips, you need to manually change `NpbWrapper.h` and recompile.