wled/WLED #485: Adding feature to support PWM driven monochannel LED strips

unimatrix27, 2019-12-19:
Hi, do you think it is possible to add the support to drive monochannel (just white) LED strips on additional GPIOs through PWM? also it would be good to have a button on a different gpio configurable, not only as button but also as switch (i do have existing hw that just has a switch)

Def3nder, 2019-12-19:
Hi @unimatrix27 ,

this is planned to be included in the next releases.
I've already implemented this into my WLED-fork https://github.com/Def3nder/WLED and opened a PR for this: https://github.com/Aircoookie/WLED/pull/481 

So stay tuned :smile:

unimatrix27, 2019-12-25:
Awesome, thanks for this. Do you know how I can change the webui html? I want to integrate a button to set a dedicated output pin to a PWM level for an external LED light, but at the same time maintaining the core function of WLED as I still have an addressable LED strip on another pin. 

I tried to use the index.html and rebuilt the C binary code with the tools mentioned in html_ui.h but I get a different binary doing this.

Aircoookie, 2019-12-28:
@unimatrix27 of course the binary changes if you change the HTML. You will need to recompile the code for your changes to the HTML to work :) Where exactly are you stuck?

closed 2020-01-04 by Aircoookie

marsic-84, 2020-02-25:
Can i use RGB ws2812 Led Strip and White Analog Strip at the same time on one ESP?? And control brightness separately?

austwhite, 2020-02-25:
@marsic-84 you are commenting on a closed issue, but to answer your question, you could use 4 individual strips and it wouldn't matter.  A combined strip is just all channels in one strip.
They just have to be the same voltage and have the commons all together.

marsic-84, 2020-02-25:
I just do not want to open another issue, but maybe i should. I think you do not understand my question. I want to use one RGB ws2812 Led Strip and one White Analog Strip, but when i set RGB type in LED setup, the white strip brightness control not available. And when i set RGBW type in LED setup,  ws2812 Led Strip becomes uncontrollable and shows strange colors (because it is not sk6812). And i need the ability to control the brightness separately, ideally through MQTT.

austwhite, 2020-02-25:
Might be best as a new issue as that is combining both digital and analogue.
Open it as a question 🙂