wled/WLED #488: mDNS only supports .local

grumpyinSD, 2019-12-19:
I don't remember the previous version not allowing me to enter my own domain suffix, however, on the latest release 0.9.0-b1 it appears it will only allow .local

Is that intentional?  I have my own DNS registered, and would like to have the devices published by my DHCP server into my local named server.  Is it possible to allow other top level domains?

In my case I use XXX.home where XXX is replaced with a name

Aircoookie, 2019-12-19:
I can also confirm that leaving off the .local works for me, but specifying a custom TLD doesn't seem to. I tried "wled.home", but neither that nor "wled.home.local" seem to work.
Are custom TLDs a part of mDNS? I always thought the .local top level domain was a given for it.

Happy holidays!

rytilahti, 2019-12-20:
mDNS uses only `.local`, other TLDs are handled as usual with regular DNS resolver. See https://en.wikipedia.org/wiki/.local (there are links to the RFCs if you want to dig deeper).

closed 2020-01-04 by Aircoookie