wled/WLED #491: Adalight baudrate limit

codamo35, 2019-12-20:
Is the baudrate for Adalight locked in at 115200? I've normally been running everything at 115200, but stretching Adalight to 400 LEDs gets quite sluggish (a few seconds of delay). I flashed a fresh binary (0.9.0-b1) at 921600 but my ESP8266 is still only accepting Adalight at 115200.
I may very well be trying to push this to it's limits... and I wouldn't be surprise if I'm just missing something, so pardon me if so! Thanks!-

Aircoookie, 2019-12-21:
Hi!
Yes, the baudrate is hardcoded to 115200. Yes, I agree, for 400 LEDs you definitely need a higher baudrate for it to run fluently. The baudrate at which you flash the binary does not change the baudrate the ESP uses to communicate with the Adalight sender, you'd need to set that in line 20 of wled05_init.ino and recompile. Did you use the standard `WLED_0.9.0-b1_ESP8266.bin`? I could send you a custom version with 921600 baud tomorrow.

Happy holidays :)

codamo35, 2019-12-21:
Hey! 
Thank you for the quick response! Well, that was simple, and it did the trick. Initially tried setting it to 921600 but it gave me a bit of noise (probably due to some outside interference). Dropped it to 460800 and it is now super responsive / smooth.
I have been thoroughly enjoying the latest release, truly outstanding work loaded into this project. Can't wait to see where it goes from here. Thanks again-

And here's to great new year! :D

closed 2019-12-21 by codamo35